ma1co/OpenMemories-Tweak #245: SOS a6300 stop recording at 35min

Pulala8, 2017-12-28:
I have the a6300 with open memories tweak intalled. When i hit rec the camera record about 35 minutes and then stops. I tried to install all the teaks back to v8.0 and the same happens every time. I have a big job coming next week and i need the camera to record over 1 hour so please if you can help me i would appreciate it. Thanks

Pulala8, 2017-12-28:
i forgot to mention my settings. I Shoot 1080p  25fps mp4

rdweaver, 2017-12-28:
I just had a similar issue with my Sony a6000 when recording in mp4. From what I was able to figure out, it was because of the file size limit with mp4. When I changed my record settings to **XAVC S** I was then able to record for hours at a time.
I would try recording in a different format and see if that works.

Pulala8, 2017-12-28:
Thanks a lot i think also that the mp4 is the problem i change it to XAVCS and works fine.

closed 2018-10-29 by ma1co