ma1co/OpenMemories-Tweak #246: Sony RX1R M2 keeps crashing after enabling Wifi in the OpenMemories-Tweak

bam2000, 2017-12-31:
Hey,
i want to use a FTP App to Transfer my bought Sony Apps from my Camera as a backup.
But I can't connect to Wifi with my FTP App, so I have to use the Open Memories-Tweak app to enable Wifi and Connect to my Wifi Router... in the app everything is working I can enable and connect to my wifi, even Telnet is working just fine, but when I try to exit the app with the menu button or the play button my camera just reboots :(

Anything I can do about it?

PS Happy new year :)

bam2000, 2018-01-01:
Okay, found an easy way to get my installed apps from the camera, adb ftw. But the problem with crashing on exit the app still exists

ma1co, 2018-01-17:
@bam2000 I don't know about the issue you're describing, but on a related note, I'd be interested in a firmware dump of the RX1RM2 :) Please tell me if you're interested.

bam2000, 2018-01-20:
Hey as long as there is no risk to my Camera, I'm in.
you just have to tell me how to dump the firmware from my Camera ^^

ma1co, 2018-01-21:
@bam2000 There shouldn't be any risk to your camera, we'll just run a command over telnet which copies the flash contents to the SD card.

Connect to your camera via telnet (i guess you figured out how this works, let me know if you need any help) and run the following command:
`dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M`
This will take some time to complete as it copies around 500MB to your SD card.

If you manage to dump the firmware as described, I'd be very grateful if you could zip it and upload it to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx). Thanks for your help!

bam2000, 2018-01-24:
Hey, wanted to tell you that I have dumped the firmware and Uploaded it to your Dropbox link :)

ma1co, 2018-01-24:
Thank you very much!

ma1co, 2018-02-09:
Here's something to try: [tweak-dev.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1711326/tweak-dev.zip) (unzip to get the apk)

The app should now exit to the android launcher instead of the menu, maybe this fixes the issue?

bulba220, 2018-02-28:
Hi,

I mainly registered to say thank you for this "dev-update"!
I've just bought an RX10M3 and put my old SD card into it. Due to its size it can get easily full. That shouldn't be a problem, because the cam has wi-fi, as well my smartphone and tablet, with about 5-5 GB free space on them. Real shame that the built in send to smartphone function sends only a low-res jpg and also no raw.
3rd party apps can't turn the wi-fi for themselves on and freeze. I had the earlier mentioned "camera restarts after exiting the OM Tweaks" problem when i tried to turn wi-fi on with that. But the latest build just works awesomely! Exiting only to the android launcher fixes the restart issue.
So if somebody wants to be able to send data from the camera to an FTP server, this is how you do it!

I'm really not into programming and electronics, but if I can help you in any ways, or just provide data, tell me.

bam2000, 2018-02-28:
> Here's something to try: tweak-dev.zip (unzip to get the apk)

The app should now exit to the android launcher instead of the menu, maybe this fixes the issue?

i Want to install the APK and test it, but all my Macs are on high Sierra and I can't connect the camera to them because the Mac will crash instantly, when I try to install the apk to the camera.

Thanks for the help maybe with some more time and a windows pc I will test it and give you feedback.

Greetings 
Carsten

ma1co, 2018-02-28:
@bulba220 Thanks for testing and reporting, I'll commit this change to the repository.
I'm always interested in firmware dumps. It would be great if you could dump the RX10M3 firmware and upload it as described in my [comment](#issuecomment-359291589) above. Please let me know if you need any help.

ma1co, 2018-02-28:
@bam2000 The os x driver situation is a mess. I can only suggest setting the camera to MTP mode, this might be more reliable.

bam2000, 2018-02-28:
thanks for the tip with MTP Mode :)
I have installed the modified apk and can confirm it is not crashing when enabling wifi on my RX1R M2
great work :)

ma1co, 2018-02-28:
@bulba220 Thanks for the dump!

closed 2018-10-29 by ma1co