ma1co/OpenMemories-Tweak #249: HDR-CX675 connection issue

efredman, 2018-01-17:
I get the following error when I try and install OpenMemories-Tweak on my camcorder. Any thoughts?

Model:              HDR-CX675
Product code:       0002445630
Serial number:      
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation HDR-CX675 is a camera in MTP mode

Switching to app install mode
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 74, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 239, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 35, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 306, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 244, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 98, in sendSonyExtCommand
  File "C:\projects\sony-pmca-re\pmca\usb\__init__.py", line 81, in _checkResponse
pmca.usb.InvalidCommandException: MTP error 0x2006

ma1co, 2018-01-17:
The camcorders are a bit complicated and I don't think anybody has managed to run an app yet. But I know more or less what to do, so please let me know if you're interested in helping the development.

* First, we have to brute force the firmware id to be able to craft custom firmware updates.
* During this process, we can dump the firmware.
* We then use updatershell to enable the app installation flag.
* We have to somehow overwrite an existing app on the camera, otherwise we won't be able to launch it.

Please let me know if you want to help!

efredman, 2018-01-17:
I'd love to help. I just bought the camcorder so as long as there isn't too much risk of bricking it, I'm happy to experiment with it.

ma1co, 2018-01-18:
Ok, let's try to brute force the model id of your camera. This won't install anything on your device, so you are pretty safe.

* Update your local repository or download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Download [0x61010000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1641136/0x61010000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Connect and power on your camera.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x61010000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

ma1co, 2018-10-29:
@efredman Feel free to re-open the issue.

closed 2018-10-29 by ma1co