ma1co/OpenMemories-Tweak #250: runMake report Execution failed for task ':app:runMake'.

tony881347, 2018-01-26:
[image]
I Do Not know how to show "$porjectDir" "$buildDir"
I just guess do make command in cmd with mingw32-make.exe
[image]
Please help

ma1co, 2018-01-27:
I'd guess that you probably don't have [arm-linux-gnueabi](https://releases.linaro.org/components/toolchain/binaries/4.9-2017.01/arm-linux-gnueabi/) installed. However, this toolchain is only used for the protection tweak binary for Japanese cameras (which runs outside the android environment). You probably don't need this binary. Try commenting out [this line](https://github.com/ma1co/OpenMemories-Tweak/blob/3f510d4127c8dc7b903d5ec5c8e9f5cdf5031844/app/build.gradle#L54) to remove it from the compilation step.

tony881347, 2018-01-31:
THANKS,I could comment out this task in gradle text,but I have another question build project:
[image]

gyfx, 2018-03-06:
@tony881347 
[https://github.com/ma1co/OpenMemories-Tweak/issues/242#issuecomment-358323354](url)