ma1co/OpenMemories-Tweak #253: Request: Enable use of XLR-K1M on A6000

motioncamera, 2018-01-31:
It seems the latest firmware for the A6000 has many of the features the A6300 has but they and disabled.   Like I can see a menu item for 4K but obviously it does not work.  So, I suspect Sony has disabled the A6000 from using the XLR-K1M audio adapter and XLR-K2M as well since I would imagine they both have the same setup.  It would be excellent to have a way around this or to enable the feature on the camera. Thanks and let me know what, I might do to begin or help in the process.  
Would this be possible through a tweak or is this a firmware issue?  Would it be something that is determined by the initial code identifying the model of the camera? Could changing that unlock the feature? I realize that there are significant hardware differences in the model from the A6300 and don't expect the 4K to work and don't care really, just want to be able to use my audio unit on both the A6500 and my two A6000s when needed. Thanks!

motioncamera, 2018-02-12:
Should I just assume there is not information, interest or has nobody done any research on the relationship between the multi-interface shoe and firmware or play memories app? If that is the case please let me know and I will cease posting and reading. Thanks

jeremyfiel, 2020-04-30:
@ma1co  any news on this capability?  Would be amazing to have XLR mic input with this adapter.