ma1co/OpenMemories-Tweak #254: Downgrade A6500 from fw 1.04+ to 1.03

AquaShooter, 2018-02-05:
Hi 

Not sure if this is where to post.  I was hoping to find a way to downgrade my A6500's FW from 1.05 down to 1.03.  FW past 1.03 have basically stopped the fast AF of my Canon lenses using the Sigma MC-11.

Please advise

Thanks!

mercier999, 2018-02-07:
I do not think there is a way to downgrade. The only solution is to buy a new body. 
I never upgrade Sony cameras unless this will solve a specific problem that affects me.

aeosdan, 2018-02-11:
Firmware can be downgraded by changing version numbers within the file that is checked and then recompressing the firmware. So for example you could make a 1.06 firmware that is actually 1.03 and flash it over the top of 1.05. I haven't done this in ages but it was successful on my A7RII.

AquaShooter, 2018-02-11:
Hi, 

That's awesome...how exactly do I do that? lol

thanks!

ma1co, 2018-10-29:
You can probably use [fwtool](https://github.com/ma1co/fwtool.py)'s _unpack_ and _pack_ commands. Be very careful! It is easy to brick a camera.

closed 2018-10-29 by ma1co

climbingkevin, 2020-07-28:
> Firmware can be downgraded by changing version numbers within the file that is checked and then recompressing the firmware. So for example you could make a 1.06 firmware that is actually 1.03 and flash it over the top of 1.05. I haven't done this in ages but it was successful on my A7RII.

Can you go into more detail how you did this? I am actually trying to do this on my A7Rii