ma1co/OpenMemories-Tweak #255: HDR CX 560V pmca.usb.InvalidCommandException

Sporky00, 2018-02-06:
Hello.
i recently have tried to install openmemories tweak on my cx 560v via the pmca gui thinking that it would work due to the software supporting the cx 700 (which i believe to be the same family) but after putting the device intro mass storage mode and trying to install the app the console in pmca gave me this error pmca.usb.InvalidCommandException: Mass storage error: Sense 0x5 0x20 0x0 i am not sure what this means but i hope i can over come it, since i brought this camera in japan ages and ages ago so it is locked in japanese and only recently have i been looking into enabling English on it. 

many thanks

ma1co, 2018-02-06:
This might be possible via a custom firmware update. As a first step, we have brute force the model id of your camera (it might be the same as the CX700, but I'm not sure about that). This won't install anything on your device. If you're interested to help, here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). If you're on os x, you have to wait for the next release.
* Download [0x11010000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1698981/0x11010000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera. Plug in the AC adapter and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x11010000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

Sporky00, 2018-02-07:
https://imgur.com/a/dOlrM
there you go, hope this helps :)
also keep up the stuff, if this works this will be a huge money saver

ma1co, 2018-02-08:
Thanks for your help, it is indead the same model id as the CX700. The CX560 is now supported in the latest development builds.

The next step is to dump the firmware, I just want to make sure that it really is the same as the CX700. Here's how to do that:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Power on your camera. Plug in the AC adapter and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console updatershell`
* Follow the instructions on your camera screen to reboot to firmware update mode. You should get to a new command prompt.
* Run the following command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following command to reboot your camera back to normal mode: `exit`
* You should now have a file called `nflasha` on your computer. Zip it and upload it to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

IMPORTANT: I had some issues with the CX700. When I last tested the above process, the camera lost all settings (same as a factory reset). I had to proceed setting the date and reconfigure all the settings in the menu. I don't know if this will happen to you as well. I'm unable to debug this, as I don't have access to the CX700 at the moment.

Sporky00, 2018-02-09:
i am having a issue with copying the firmware, i leave it to copy for around 1 - 2 hours and after 40ish minutes the camera shuts off and disconnects from my pc even tho i have it plugged into ac :( i have tried to copy 3 times already all end with failure :( i will also be over seas for the next two days so i can not do anything until Monday, sorry

ma1co, 2018-02-09:
The process should take maybe a minute or two, but not half an hour. Maybe you missed the completion? Have a look at the folder where you donwloaded _pmca-console.exe_. There should be a file called _nflasha_. What's its size? If it is around 128MB, everything completed successfully.

Sporky00, 2018-02-11:
should be uploaded, idk why it didnt work but i changed computers and i got it finally.

ma1co, 2018-02-11:
Thank you, it is indeed almost the same firmware as the CX700.

Here is what you can try to unlock all languages (I don't know if this will work):

* You can use pmca-gui for this, download the latest development build [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) (alternatively, you can use the `tweak` command in updatershell).
* In pmca-gui, go to the _Tweaks_ tab and click the _Start tweaking_ button. The camera should reboot to updater mode.
* Check _Unlock protected settings_
* Check _Unlock all languages_
* Uncheck _Unlock protected settings_
* Click _Done_. Your camera should reboot to normal mode.
* Go the the language selection menu. With a bit of luck, you should have all options now.

Please let me know if it worked.

Sporky00, 2018-02-11:
https://imgur.com/a/SxJx2 :(

ma1co, 2018-02-12:
There seems to be a communication error. Which computer are you using? Is it the one you had issues with dumping the firmware?

Sporky00, 2018-02-12:
nope i am using my mac book pro with bootcamp, the one that dumped the firmware, i will try to tweak with all the pcs i have available to me, thanks for you help i should hopefully be able to get everything working!

ma1co, 2018-02-14:
I just discovered that this is not a communication issue, it's a problem in Sony's Backup_write() code which causes the camera to lock up. My bad. I'll try to fix this by changing the Backup_guess_protection() code.

ma1co, 2018-02-16:
The issue should be fixed in the newest development builds.

Sporky00, 2018-02-20:
sorry for being inactive, i was again on another trip but thank god everything works now! i cant thank you enough for helping me. i wish you good luck on this and your future projects :)

ma1co, 2018-02-21:
No worries, thanks for your help on the project!

closed 2018-02-21 by ma1co

TimSchimansky, 2022-11-19:
@ma1co I get the exact same error: `Mass storage error: Sense 0x5 0x20 0x0` for my Sony QX10 LensCamera. The firmware of that device is not found by the guess function. Therefore i have uploaded the firmware into your Dropbox.