ma1co/OpenMemories-Tweak #257: prevent USB mode screen from showing up to allow camera use while charging a6000

aaronwi, 2018-02-11:
This would be a much needed feature to allow for external battery packs and AC adapters to keep the camera charged and allow it to take photos, I believe the a6300 or a6500 allow for camera use while charging, so their firmwares could be reference points.

aeosdan, 2018-02-11:
This would be a great feature but I don't know if we have access to the necessary part of the code. Sony could have enabled this for many older cameras through firmware updates but have not done so. That points to two potential explanations. Either Sony don't care about adding useful features and would rather sell you a new camera, or they are unable to do this due to hardware implementation.

mturkdfeldt91, 2018-02-12:
I read somewhere that this is not possible on the a6000 because of the circuitry. Not sure if it is true.

I suggest a dummy battery and a battery made of 4 or 6 18650 cells for a bike headlight. You can get the battery packs for about $10-20 usd and the dummy battery is about the same.