ma1co/OpenMemories-Tweak #260: [REQEUST] Longer than 30s exposure

rawhead303, 2018-02-19:
I thought this request would already be here, but a search didn't hit anything.

Any way you could remove the 30s limit to long exposure?  Would love to see up to 20 minutes :D

AlucardleVash, 2018-02-26:
You certainly already know that, but you can use a remote for that managing your bulb mode ?

rawhead303, 2018-02-26:
Of course, that's what I do.  But it's great to have it in-camera, because a remote is one more thing that can get lost, forgotten at home, or run out of battery.

bcdub, 2018-03-11:
This might fix the “star eater” issue on the a7/r/s. This would be huge. +1

qualitywins, 2018-11-27:
Hi,

I think this has hardware limitations. Non Bulb exposure times are controlled by the CMOS itself usually.  If the base timing is 500 KHz and the timer implemented on the CMOS has a depth of 24 bits it would limit the exposure to 33.55 Sec.

That's also why webcams cannot do long exposures, the depth of the timer is even a lot smaller, and they don't support external timing, e.g. "bulb".

I think the only way to get rid of the star eater would be to identify a config parameter (if it exists) and to switch it off.

Kind regards,

Nikolas

rawhead303, 2018-11-27:
But even in that case, if an external remote can do it, surely there should be a way to control the camera to shoot in Bulb for a set number of seconds?    I guess that might be more along something to be achieved via an app rather than firmware...