ma1co/OpenMemories-Tweak #261: Sony rx1r (not rx1rm2) japan edition language change

sporre, 2018-02-28:
Hello,

There was already a previous similar ticket but it was closed already. I have a Sony RX1R bought in Japan (not mk2) I want to know if it is also possible to change the language on this camera. Now it's only in Japanese. It appeared to be possible but I have no clue how. 

Thanks!

ma1co, 2018-02-28:
This should be possible via a custom firmware update. As a first step, we have brute force the model id of your camera. This won't install anything on your device. Here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). If you're on os x, you have to wait for the next release.
* Download [0x22020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1766273/0x22020000.zip) and [0x32020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1766275/0x32020000.zip) and place them in the same folder (no need to unpack). They contain a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following commands: `pmca-console guess_firmware -f 0x22020000.zip` and `pmca-console guess_firmware -f 0x32020000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

Richiard, 2018-03-09:
Wow!!! What a SUPER DUPER CLEVER you are!!!! 
Can I ask you something?
Does the firmware have German Menu?
THX A LOT in advance!
Richiard.

ma1co, 2018-03-27:
@sporre v0.17 has been [released](https://github.com/ma1co/Sony-PMCA-RE/releases/latest).

sporre, 2018-03-29:
Have download the new version. And also the two zip files. I place everything in one folder in “Applications”

I’m having difficulty running the command. Since literally copying your words does not give me any feedback (sorry, I’m new to programming)..

If you could tell me what steps to take, that would very helpful.

Sorry for the inconvenience.

Thanks!

Alexander

ma1co, 2018-03-29:
@sporre 
* Open a terminal
* Navigate to the directory: `cd path/to/Applications`
* Make the file executable: `chmod 755 pmca-console-v0.17-osx`
* Execute it: `./pmca-console-v0.17-osx guess_firmware -f 0x22020000.zip`

ma1co, 2018-04-04:
Thank you, the RX1R is now supported in updatershell.

Since there is no new OS X build, please download [RX1R.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1875450/RX1R.zip) and unpack it to the folder containing pmca-console.

The next step is to dump the firmware. Here's how to do that:

* Connect the camera.
* Run `pmca-console updatershell -f RX1R.dat`
* The camera should reboot to updater mode.
* Run the following updatershell command to dump the firmware: `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot1` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

ma1co, 2018-04-12:
@Richiard Thank you for the upload. Unfortunately, the _nflasha_ file is corrupt. It should be about 250MB. Can you please try again?

Richiard, 2018-04-13:
@ma1co 

Dear, Ma1co,

It shows that Error: (31, 'DeviceloControl.... Device doesn't work...)
Maybe, do you have an idea why it doesn't work?!

ma1co, 2018-04-13:
When does it occur? Sometimes you can ignore it and retry...

closed 2018-10-29 by ma1co

damarogit, 2019-05-25:
Hi ma1co, 
I have the same issue with the same camera RX1R(var. I).
Now is having only Japannese and I would like to install English in order to use it.
Can you help me with that, please ?
Thank you so much,