ma1co/OpenMemories-Tweak #263: Tweak did not work on DSC-HX90V

gkln, 2018-03-01:
HX90 (HX90V) is not listed at the list of supported cams, but HX80 is. Surprisingly, Tweak didn't work. It recognizes correctly the cam, but when "start tweaks" is clicked, "Unknown device: DSC-HX90V" is displayed. 
After successfully tweaking ILCE-9100, RX100M5 and WX220, I think the problem is not my installation. 
Repetition of the process after restart the pmca-gui---exe didn't change the behavior.

Environment
W732EN, all updates.
pmca-gui-v0.16-win.exe

Messages:
======================
Loading app list
Found 9 apps

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX90V is a camera in MTP mode

Getting device info
Using firmware for model DSC-HX90V

Unknown device: DSC-HX90V
======================================

ma1co, 2018-03-01:
The HX90V is present in the [list of cameras supporting android](https://github.com/ma1co/OpenMemories-Framework/blob/master/docs/Cameras.md). This means that you can go to the "Install app" tab in pmca-gui, choose "OpenMemories: Tweak" and click "Install". On your camera, go to the application list, open the app and apply the tweaks there.

The HX90V is currently not supported in firmware update mode, this is because we don't have a firmware dump of the camera. Please let me know if you're willing to dump the firmware using telnet, I'll provide you with instructions. Your help would be much appreciated.

gkln, 2018-03-01:
Of course I am willing, just I need detailed step-by-step instructions.
Many thanks for your great work!

ma1co, 2018-03-01:
Ok, here's what you can do:

* Install the "OpenMemories: Tweaks" app as described above.
* Open the app, go to the developer tab and enable wifi and telnet. Use the wifi settings button to connect to your wifi. Under "Enable wifi", you should see the IP address of your camera.
* Connect your computer to the same wifi network and download [putty](https://www.putty.org/).
* Open putty, enter the camera's IP address and select "Telnet", click "Open". You should see a command prompt which allows you to execute commands on your camera.
* Enter the following command: `dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M` and hit enter.
* The camera will dump the firmware to the SD card. Wait until the card led stops flashing (it may take several minutes).
* Please zip the DUMP.DAT file from the SD card and upload it to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Please let me know if you need any help.

gkln, 2018-03-01:
Well, first problem: cannot connect to my own network.
From your app: after the first try and the msg, I should again key in my PW (26 lower and uppercase letters) when trying so the "keyboard" didn't show up any more (bug?). And, by the way, to hide the PW is such a situation is no good idea - no possibility to check finally if all was keyed in exactly. And it is hard work to key in 26 mixed case signs and digits.
So I tried again with the cams basic WIFI-function the setup - also failed. The keyboard is here different, the method of hiding the characters similar. 
This is an intermediate report - I will keep on trying to set up a WiFi connection to my home network. 
BTW: is there no possibility to perform those task by using an USB connection?

ma1co, 2018-03-01:
Yes, there is also a way to do it over USB. As a first step, we have brute force the model id of your camera:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Download [0x51020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1772300/0x51020000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x51020000.zip`
* With a bit of luck, the script should find the model id.

Let me know if you need any help.

gkln, 2018-03-01:
I succeedet to connect to my homenet, then changed to your app. I opened putty, i got a window connected to my cam, but was unable to copy in the command you posted. when i tried manually, it also didnt work. in the meantime the network connection timed out, and, returning to the cam, it looks the cam didn't remember the PW!!!
Give me a little break. Afterwards I will try the USB-method.
I have GMT+1

ma1co, 2018-03-01:
For the app method, make sure you set the camera's auto power off time to 30min. It should remember the password and connect back to the wifi when you re-enable wifi and telnet in the app. You can paste in putty by using the right mouse button.

gkln, 2018-03-01:
well ...  I had a little to adopt the command to the situation. However:
----
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-9-geef14a3-win.exe guess_f
rmware -f 0x51020000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX90V is a camera in MTP mode

Getting device info
Model name: DSC-HX90V

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen3_0x51020037_0.dat

ma1co, 2018-03-01:
Thank you, the HX90V is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Connect the camera.
* Run `pmca-console updatershell` and your camera will reboot to firmware update mode. You should see a command prompt on your computer.
* Run the following command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following command to dump the bootloader partitions: `bootloader`
* Run the following command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot1` to `boot5` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Let me know if you need any help.

gkln, 2018-03-01:
Fail ...

Maybe it is in connection with the initial state of the camera, being connected via USB and powered on: at first there is a question to change the mode. I set there no action bc. not sure, which is the right on. Also later "Please follow the instructions on the camera screen." I am not sure which action is appropriate.
Here is the output from the cmd-prompt:
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX90V is a camera in MTP mode

Getting device info
Using firmware for model DSC-HX90V

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-console.py", line 96, in <module>
  File "C:\projects\sony-pmca-re\pmca-console.py", line 82, in main
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 326, in updaterShel
lCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 365, in firmwareUpd
ateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 337, in firmwareUpd
ateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 519, in init
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 466, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 74, in sendSonyExtComma
nd
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 227, in s
endReadCommand
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 206, in _
sendScsiCommand
pywintypes.error: (31, 'DeviceIoControl', 'Ein an das System angeschlossenes Ger
ät funktioniert nicht.')
[5664] Failed to execute script pmca-console

E:\Sw\HW_Treiber\Camera\SonyTweak>

ma1co, 2018-03-01:
I can't really tell what's going wrong. As far as I can see, you're doing everything right (you don't have to do anything on the camer, it should reboot automatically). It should work the same as the WX220. Maybe try killing PMH by stopping the "Media Check Tool" task in task manager...

gkln, 2018-03-01:
Killed process PMBVolumeWatcher.exe
Connected cam and switched on. This time Window comes up with proposals to import media. Ignored.
Rerun the commands from command prompt. 
After commiting the only option on the cams screen, is executed the indicated command.
Here is the cmd-console output:
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX90V is a camera in MTP mode

Getting device info
Using firmware for model DSC-HX90V

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen. <==== on the cam screen is written: "Follow  computer instructions" - then the screen turns to black.

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-console.py", line 96, in <module>
  File "C:\projects\sony-pmca-re\pmca-console.py", line 82, in main
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 326, in updaterShel
lCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 365, in firmwareUpd
ateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 337, in firmwareUpd
ateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 519, in init
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 466, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 74, in sendSonyExtComma
nd
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 227, in s
endReadCommand
  File "C:\projects\sony-pmca-re\pmca\usb\driver\windows\msc.py", line 206, in _
sendScsiCommand
pywintypes.error: (31, 'DeviceIoControl', 'Ein an das System angeschlossenes Ger
ät funktioniert nicht.')
[5664] Failed to execute script pmca-console

E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX90V is a camera in MTP mode

Getting device info
Using firmware for model DSC-HX90V

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...

Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /dev/nflasha
Writing to .\nflasha...
>exit
Done

E:\Sw\HW_Treiber\Camera\SonyTweak>

Terminated afte about 6-7 minutes.
After some search I found nflasha with 595592 kb in the pmc-console-directory, but NO boot1 - boot5.
Should those boot1-... files reside in the same directory?
Does it make sense to upload only the nflasha-file?

ma1co, 2018-03-01:
Great, so it worked! Don't worry about boot1 - boot5, that's just the bootloader, it's not very important (you'd have to run the `bootloader` command to create these files). The nflasha file is the important firmware file. Please zip the nflasha file (this makes it smaller, so the upload is faster) and upload it to the dropbox. Thanks!

Just for your information (if you're curious): If you wonder what's inside the nflasha file, you can extract it using [fwtool](https://github.com/ma1co/fwtool.py) (see the readme for instructions).

gkln, 2018-03-01:
Please check if the file uploaded correctly. The dropbox-window doesn't say anything if the upload terminated with rc=0.

ma1co, 2018-03-01:
I received the file, everything is fine. Thank you very much, your help is much appreciated!

closed 2018-03-01 by ma1co

maeotaku, 2019-12-14:
@gkln did you ever made HX-90V work? i was trying today with no luck, GUI does not identify the camera at all :(

maeotaku, 2019-12-14:
@gkln nevermind just worked! thanks both for enabling support for this model :)

gkln, 2019-12-14:
Yes, it worked for RX100M5, HX90V, ILCE-5100. But it is more than one year ago I installed the tweak and I cannot remember the details any more. Maybe for identifying the camera you have first to change some settings in the USB-setup of the camera. Try all possibilities there one after another. I can guarantee in my case that finally I got the tweak installed successfully.

RogerBlue, 2021-03-03:
Has anyone had any luck with any of these methods on the DSC-HX99?

gkln, 2021-03-04:
Yes, in my case it was successful.

RogerBlue, 2021-03-04:
> Yes, in my case it was successful. Von: RogerBlue [mailto:notifications@github.com] Gesendet: Mittwoch, 03. März 2021 10:34 An: ma1co/OpenMemories-Tweak Cc: gkln; Mention Betreff: Re: [...]

On a DSC-HX99, not the DSC-HX90V you mentioned?

gkln, 2021-03-04:
Sorry, you are correct: my cam is a DSC-HX90V, not a HX99. I made two mistakes: first, I thought the “HX99” in the body of your mail was a mistype of “HX90V” as written in the subject line, and then I thought that the last letter is of no importance. Sorry again. 
In contrast to the subject line I can confirm: the tweak worked (in the meantime, already long ago) for my DSC-HX90V.