ma1co/OpenMemories-Tweak #265: Testing HDR-XR550V

gkln, 2018-03-01:
Was curious if it works also on a handycam and it would have been fine if it would be possible to enable also all languages - so I gave it a try.

Well, here is are the 
messages
===================
Loading app list
Found 10 apps

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Model:              HDR-XR550V  
Product code:       0002061430
Serial number:      00120883
Firmware version:   1.01
GPS Data:           2018-03-01 00:00:00 - 2018-03-31 00:00:00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Getting device info
Using firmware for model HDR-XR550V  

Unknown device: HDR-XR550V  
=============================

ma1co, 2018-03-01:
The HDR-XR550V was announced in 2010, it is too old to apply any tweaks (it is probably a gen0 or gen1 camera, they feature a different settings system as newer models, I didn't spend any time trying to develop tweaks for it).  However, it would still be interesting to dump the firmware and maybe run some code on it. There is probably a lot of guessing involved, since we don't know anything about that model. But if you're willing to help, I can provide you with instructions. The first step would be to find out which kind of firmware image it uses.

gkln, 2018-03-01:
Supply me with the detailed instructions, I will try my best.
Also indicate how great (MB) the dump will be - my connection is not that fast so maybe I need a little planning at which time I can perform the transmission.

ma1co, 2018-03-01:
Ok, let's find out if it is a gen0 or a gen1 camera. Please download this [zip file](https://github.com/ma1co/OpenMemories-Tweak/files/1772278/gen.zip) and unzip it. Open pmca-gui, connect your camera and go to the "Update firmware" tab. Select gen0.dat and click "Update firmware". Then select gen1.dat and click "Update firmware". Please post the console output.

gkln, 2018-03-02:
Here is the console-log:

Loading app list
Found 10 apps

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Model:              HDR-XR550V  
Product code:       0002061430
Serial number:      00120883
Firmware version:   1.01
GPS Data:           2018-03-02 00:00:00 - 2018-04-01 00:00:00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

ma1co, 2018-03-07:
Thanks for testing and sorry about my delayed reply. So the camera doesn't seem to use a gen0 or gen1 firmware image. I have two more files to test. Please repeat the process described above with this zip file: 
[gen2.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1791089/gen2.zip)

gkln, 2018-03-08:
I tried once more without choosing an option. In this situation the msg "No devices found. Ensure your camera is connected." is displayed as soon as I click on "get camera info".

Then I repeated the procedure with the option "USB-Connect to the SDLC-Card". Same result as with HDD.

Here is the log:
gen0, gen1, gen2, and finally m5.dat

Loading app list
Found 10 apps

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
No devices found. Ensure your camera is connected.

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Model:              HDR-XR550V  
Product code:       0002061430
Serial number:      00120883
Firmware version:   1.01
GPS Data:           2018-03-07 00:00:00 - 2018-04-06 00:00:00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 98, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 300, in firmwareUpdateCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 339, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 522, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 496, in _sendWriteCommands
Exception: Firmware update error: Invalid data

ma1co, 2018-03-08:
I don't think the USB mode is the issue. It really seems that this camera is using different firmware encryption keys. I'll have to think about whether it is viable to brute force the complete firmware image...

gkln, 2018-03-08:
In my eyes it is not a very big problem. As you stated, the cam was announced 2010 AND my model offers at least English as menu language, which is already a big thing compared with Japanese only.
(If I could choose: I would prefer if your brute-force efforts concentrate rather on the  RX10M4 ;) )

ma1co, 2021-08-12:
pmca-console now supports "senser" mode, the USB service mode present on all cameras I know of. It should be possible to dump the firmware of these unsupported cameras using this mode. If you want to help, here is what you can try:

* Clone or download the [Sony-PMCA-RE](https://github.com/ma1co/Sony-PMCA-RE) repository, or, on Windows, download the newest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) of pmca-console.
* If you are on Windows, follow the [instructions](https://github.com/ma1co/Sony-PMCA-RE/#windows-drivers) to install the USB drivers.
* Connect the camera in mass storage mode and run `pmca-console serviceshell`. If everything works out, the camera should switch from the USB screen to the playback / recording screen and your computer should show a command prompt.
* Type `pull /dev/nflasha` and wait for a while for the command to complete, then type `exit`.
* If everything works, you should have a firmware dump in a file called "nflasha". Please zip it and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Please let me know if you need any help.

gkln, 2021-08-12:
Hi, thanks for your support!
 
These were my steps:
Download pmca-console-v0.17-63-g77b2991-win
zadig-2.5
 
Connecting camera Sony HDR-XR550V with USB-Cable to PC with W732 EnterpriseEdition.
The display of the camera immediately changes und offers the following possibilities (see Attachment Sony HDR-XR550V with USB-Cable to PC with W732 EnterpriseEdition.HEIC (the new format of iPhone Xs; in case of troubles I can convert it to jpg)
I selected the left one with the HDD symbol, as the camera has not SDLC-Card in the slot and I expect the camera-SW to be stored on the internal HDD
After pressing the left USB Connect the next screen looks 
20210812_2 Sony HDR-XR550V with USB-Cable to PC with W732 EnterpriseEdition, connected.HEIC
Run Zadig-2.5.exe
Selected the Cam HDR-XR550V and right side as indicated libusb-win32 and replaced driver.
 
The process terminated OK, driver replaced.
Now instructions on  <https://github.com/ma1co/Sony-PMCA-RE/#windows-drivers> GitHub - ma1co/Sony-PMCA-RE: Reverse Engineering Sony Digital Cameras say_
*	“Run pmca-console serviceshell to make the camera switch modes.”
I run “Download pmca-console-v0.17-63-g77b2991-win.exe”, being not sure if this is also the “serviceshell”
However, within some seconds the process terminated, an intermediate window that could have been a command prompt disappeared.
 
*	Once the camera has switched, repeat the above step to install a driver for service mode.
Absolutely unclear what this means! However, the display of the videocam didn’t change at all from this procedure. Did the camera switch? I have no idea.
Next step: “repeat the above step to install a driver for service mode.” ????? 
I again selected my cam as shown in the screenshot above, and having no further information selected again the libusb-win32 driver. No idea, if this is correct. “install a driver” for me is no clear instruction.
 Screenshot shows the second replacement.
 
 
Next I run again pmca-console-v0.17-63-g77b2991-win.exe, but same effect as before. 
According to your email I should connect (again?) the camera in mass storage mode. So I ended the usb-connection, disconnected the usb-cable, reconnected it.
Same screens appeared as in attachments shown.
Rerun pmca-console-v0.17-63-g77b2991-win.exe, but again no change on the display of the cam neither an open command window. 
Additional remark: I run the pmca-console-v0.17-63-g77b2991-win.exe from the windows file manager by double-click. If it should be started from a command-prompt, maybe with admin rights, please tell so.
 
Videocam is still working ;)
 
 
Regards, Gerhard

 

Windows Drivers
To use service mode on Windows, custom drivers have to be installed using Zadig <http://zadig.akeo.ie/> :
·         Make sure the camera is connected in mass storage mode.
·         In Zadig, check Options -> List All Devices, select the camera, select libusb-win32 and click Replace Driver.
·         Run pmca-console serviceshell to make the camera switch modes.
·         Once the camera has switched, repeat the above step to install a driver for service mode.
·         You should now be able to use pmca-console serviceshell.
To be able to use the camera normally again, the libusb drivers have to be uninstalled in device manager.
Browser Compatibility
The browser-based installer can be found at sony-pmca.appspot.com <https://sony-pmca.appspot.com/> . This site uses the official Sony browser plugin to communicate with the camera directly from a browser window. Since other browser vendors are disabling NPAPI plugins, this method only works in Internet Explorer and Safari. Additionally, camera firmware updates released in June 2017 and later explicitly whitelist the URL of the official app store. Updated cameras refuse to install apps from our site. It is thus recommended to use the native installer instead.
Meanwhile, the site is still used to keep track of the installation counters for the apps in the app list.
Zadig - USB driver installation made easy (akeo.ie) <https://zadig.akeo.ie/>  :
[Zadig icon] Zadig
USB driver installation made easy
[Zadig screenshot]
Zadig is a Windows application that installs generic USB drivers, such as WinUSB <https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb> , libusb-win32/libusb0.sys <https://sourceforge.net/p/libusb-win32/wiki/Home/>  or libusbK <http://libusbk.sourceforge.net/UsbK3/> , to help you access USB devices.
It can be especially useful for cases where:
you want to access a device using a libusb-based application
you want to upgrade a generic USB driver
you want to access a device using WinUSB
Note: "libusb-based" above means an application that uses either libusb <https://libusb.info/> , libusb-win32 <https://sourceforge.net/p/libusb-win32/wiki/Home/>  or libusbK <http://libusbk.sourceforge.net/UsbK3/> .
Download
Updated 2020.03.28:
Zadig 2.5 <https://github.com/pbatard/libwdi/releases/download/b730/zadig-2.5.exe>  (4.9 MB)
Other versions <https://zadig.akeo.ie/downloads/> 
System Requirements:
Windows 7 or later.
Windows XP and Windows Vista are NO LONGER SUPPORTED.
Usage
Download the executable and run it — no installation is necessary.
If elevation is required, you will be prompted for it.
An usage guide for Zadig is available HERE <https://github.com/pbatard/libwdi/wiki/Zadig> .
The executable is digitally signed and the signature should state: "Akeo Consulting"
Frequently Asked Questions (FAQ)
A Zadig FAQ is available HERE <https://github.com/pbatard/libwdi/wiki/FAQ#wiki-Zadig> .
To provide feedback, report a bug or request an enhancement please use the github issue tracker <https://github.com/pbatard/libwdi/issues> . Or you can send an e-mail ***@***.***?subject=Zadig> .
 
·         Original geht hier weiter

ma1co, 2021-08-12:
@gkln You are supposed to run it from the command prompt like this:
`pmca-console-v0.17-63-g77b2991-win.exe serviceshell`

gkln, 2021-08-12:
But camera has switched!
 
Now the camera disappeared, but a Sony USB Device is available. I took this.

 

 
 

 

 
 

 
The resulting „nflasha“ has 0 kB!
Please check console-output: I thought, as soon as I see the “>” sign, I can execute exit.
Anything else wrong?

ma1co, 2021-08-12:
Nice, that's good progress! I remember having trouble dumping /dev/nflasha on my NEX-3, too. Can you try this instead?
`pull /usr/lib/libupdaterapi.so`
This should contain the firmware keys.

gkln, 2021-08-12:
Success so far, 143 kB file.
I zipped it and upload to your site as 
20210812_184800 HDR-XR550V libupdaterapi.zip
 
Hopefully my cam and my PC still working. Unsure about uninstalling the libusb-win32 (v1.2.6.0) driver and return to original that was 
USBSTOR (v6.1.7601.23403)
 
cu, gkl

gkln, 2021-08-12:
> Very nice, we have the firmware keys now.
> My next step is to create custom firmware file, so we are able to use updatershell (which has more functionality than serviceshell, at least for now). For that, we need the model id, which is stored on nflasha1. So let's try to dump this partition.

Failure!
 
Restart

 
 

 

 
 
command inserted with copy - paste
 

 

 
no output. looks as „dd“ is not known to the established linux shell.
Should I provide you with a teamviewer-address and you can try yourself?

·         Type shell. You should now see a linux shell.
·         Execute dd if=/dev/nflasha1 of=/tmp/nflasha1
·         Once this has completed, press <CTRL+C> and then to go back to the serviceshell prompt
·         Execute pull /tmp/nflasha1
You should now have a file called nflasha1.

ma1co, 2021-08-12:
I cannot see your screenshots on github. Can you copy-paste the output instead?

gkln, 2021-08-12:
Begin output in shell ================================
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe
service
shell
Using drivers libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Switching to senser mode
 
Waiting for camera to switch...
Found a camera in senser mode
Authenticating
Welcome to USB service shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>shell
Terminal activated. Press <CTRL+C> to exit.
 
 
BusyBox v1.4.2 (2009-01-20 14:59:28 JST) Built-in shell (ash)
Enter 'help' for a list of built-in commands.
 
/bin/ash: can't access tty; job control turned off
/ # dd if=/dev/nflasha1 of=/tmp/nflasha1
/bin/ash: dd: not found
/ #
End output in shell ================================

 

<https://github.com/ma1co/OpenMemories-Tweak/issues/265#issuecomment-8978626
QFRVANCNFSM4ETB5BDQ> .
t=8&pt=524675>  or Android
gn=notification-email> .
<https://github.com/notifications/beacon/AIZ6V7I7PG6SVTSVQCM25Q3T4QFRVA5CNFS
M4ETB5BD2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGWCEYHA
.gif>

ma1co, 2021-08-12:
Can you try this instead?
`/bin/busybox dd if=/dev/nflasha1 of=/tmp/nflasha1`

If this does not work, we can also mount the partition.

karak21, 2021-08-12:
Hey ma1co, I tried gaining a service shell but it said access denied and now it says there are no sony devices anymore even though I am connected to my camera

karak21, 2021-08-12:
Hey gkln, any idea why my camera does not want to connect?

ma1co, 2021-08-12:
@gkln This is just log output from the camera. You can press enter to get a new console prompt.
The environment seems to be quite a bit different from my NEX-3. Can you try this?
`/bin/busybox dd if=/dev/nflasha1 of=/var/nflasha1`
and later
`pull /var/nflasha1`

ma1co, 2021-08-12:
@karak21 Which operating system are you using? Can you share the steps that you have taken and the exact output that you got?

gkln, 2021-08-12:
I have no idea! Maybe a good idea to restart the process from the beginning:
 
1.      Disconnect camera and power off
2.      camera power on and reconnect with usb-cable
3.      In my case immediately the display of the camera changes. According to instructions I should change to mass storage mode, but in my case HDR-XR550V this is not offered. I selected USB on the left hand with the symbol of the harddiskdrive HDD.
4.      Running zadig-2.5.exe by double-click from the windows file manager.
5.      Open options-menu and mark “list all devices”
6.      select your cam – in my case it shows up in de list
7.      on the right hand select the libusb-win32 (v1.2.6.0) and click “reinstall driver”
8.      Open a cmd-prompt as administrator and CD into the directory with the downloaded pmca-console-v0.17-63-g77b2991-win.exe 
9.      Execute at the commandprompt > pmca-console-v0.17-63-g77b2991-win.exe serviceshell
10.  The camera should switch to recording-mode or so
 
maybe this helps. There is some uncertainty bc. zadic mentions to be run twice.

gkln, 2021-08-12:
Failure: cannot find pull-cmd
Console output========================
←[0m875.2438s 02A136:0365 141 33 03 3 ←[1m[FM_MPB]Evt:TransOff,
ULL, Act:FFFFFFFF, Path:←[0m
←[0m875.2440s 02A136:0386 141 33 03 3 ←[1m[MODE-BASE]Notify to A
(0x7402140)←[0m
Please press enter
 
>shell
Terminal activated. Press <CTRL+C> to exit.
 
 
BusyBox v1.4.2 (2009-01-20 14:59:28 JST) Built-in shell (ash)
Enter 'help' for a list of built-in commands.
 
/bin/ash: can't access tty; job control turned off
/ # /bin/busybox dd if=/dev/nflasha1 of=/var/nflasha1
/bin/busybox dd if=/dev/nflasha1 of=/var/nflasha1
6144+0 records in
6144+0 records out
/ #
/ # pull /var/nflasha1
pull /var/nflasha1
/bin/ash: pull: not found
/ #
End console output =====================

gkln, 2021-08-12:
> @gkln <https://github.com/gkln>  Exit from the Linux shell with CTRL + C and Enter before running the pull command (as described a few posts above).

Worked!
 
File uploaded as „20210812_211400 HDR-XR550V libupdaterapi.so nflasha1.zip”
 
Start console output ======================
ULL, Act:FFFFFFFF, Path:←[0m
←[0m451.3095s 04123B:08EF 141 33 03 3 ←[1m[MODE-BASE]Notify to APP Final-TRGEVT
(0x7402140)←[0m
Please press enter
 
> 
>pull /var/nflasha1
Writing to .\nflasha1...
> 
End console output ======================

gkln, 2021-08-12:
> Great, now we are finally done with serviceshell. You can uninstall the usb drivers:
> Open the Windows device manager, check View -> Show Hidden Devices, then uninstall the drivers for the two devices under "libusb".
> Now we should be able to use updatershell to finally dump the full firmware:
> ·         Download HDR-XR550V.txt <https://github.com/ma1co/OpenMemories-Tweak/files/6977999/HDR-XR550V.txt>  and place it in the same directory as pmca-console.
> ·         Re-connect your camera and run pmca-console updatershell -f HDR-XR550V.txt. You might have to press a button to restart the camera to updater mode.
> ·         You should now be greeted by another shell.
> ·         Run pull /dev/nflasha
> ·         Run bootloader
> ·         Run bootrom
> ·         Run exit to restart the camera
> Please note that the camera may be initialized to factory settings during this process.
> We should now have the different firmware files: bootrom, boot0 and nflasha.

Failure on  run pmca-console updatershell -f HDR-XR550V.txt  ==> No devices found. Ensure your camera is connected.
 
Start Console output ========================
...
> 
>exit
Done
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell -f hdr-xr550v.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
No devices found. Ensure your camera is connected.
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End Console output ========================
 
Camera-Display says: “Connected. Do not unplug USB ...”

ma1co, 2021-08-12:
@gkln Have you uninstalled the libusb drivers in device manager? And maybe disconnect and reconnect the camera.

gkln, 2021-08-12:
Yes, in the sequence you indicated:
 
Quote: ===========
Open the Windows device manager, check View -> Show Hidden Devices, then uninstall the drivers for the two devices under "libusb".
Now we should be able to use updatershell to finally dump the full firmware:
*	Download  <https://github.com/ma1co/OpenMemories-Tweak/files/6977999/HDR-XR550V.txt> HDR-XR550V.txt and place it in the same directory as pmca-console.
*	Re-connect your camera and run pmca-console updatershell -f HDR-XR550V.txt. You might have to press a button to restart the camera to updater mode.
*	You should now be greeted by another shell.

ma1co, 2021-08-12:
Then it should work if you reconnect the camera. Does it show up in device manager / explorer / zadig?

gkln, 2021-08-12:
Disconnected and reconnected several times. Windows gives the usual sound-signal on finding a new device. In the system tray it said new device found, installing drivers.
 
In the windows explorer cam doesn’t show up, in Zadic it does, but with driver “libusb0 (v1.2.6.0)” (originally “USBSTOR V6.1.7601.23403”)
In device mgr there is now an entry 
libusb-win32.devices
            CIF Single Chip
            HDR-XR550V
Though I uninstalled them. CIF had before also another driver. 
 
I uninstalled the driver another time – but no reboot, as I have other work to do.
 
Reconnecting the CAM installs driver, the display of the cam says “Preparing”
Windows reports driver for HDR-XR550 successfully installed.
but in device mgr again it shows up in the section “libusb-win32 devices”
Cam display says now “connected”, but cam still not visible in windows explorer, but as before in Zadig.

ma1co, 2021-08-12:
It seems that Windows keeps finding the libusb driver. When uninstalling it in device manager, can you check "delete the driver software for this device"?

gkln, 2021-08-12:
I checked this time this box at uninstall.
Cam reconnected – searches and finds driver.
Cam switched to USB-mode: one more driver installed by windows. 
In devmgr, section drives, Sony “Camcorder USB Device” shows up.
Libusb-section not there any more.
Zadig shows HDR-XR550V
Windowsexplorer lists a exchangeable drive.
 
Resuming the proposed process: Failure:
Start Console output ==================
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell -f hdr-xr550v.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Traceback (most recent call last):
  File "pmca-console.py", line 100, in <module>
 File "pmca-console.py", line 84, in main
  File "pmca\commands\usb.py", line 377, in updaterShellCommand
  File "pmca\commands\usb.py", line 390, in firmwareUpdateCommandInternal
  File "pmca\usb\sony.py", line 557, in checkGuard
  File "pmca\usb\sony.py", line 531, in _sendWriteCommands
Exception: Firmware update error: Unknown (0x102)
[7936] Failed to execute script 'pmca-console' due to unhandled exception!
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End Console output ==================

ma1co, 2021-08-12:
Not sure what error 0x102 means, but it might be related to the battery:
```
 STAT_LOW_BATTERY = 0x100
 STAT_HOT_BATTERY = 0x101
 STAT_AC_ADAPTER_REQUIRED = 0x103
 ```
 Can you plug in the camera through the AC adaptor? And if this does not work, try fully charging the battery.

gkln, 2021-08-12:
The cam is operated the whole time on the AC power supply in addition to a fully charged battery. Disconnecting the power supply shows 390 min operational time for the battery

ma1co, 2021-08-12:
Do you get the same error when the AC adapter is disconnected?

gkln, 2021-08-12:
Steps:
Camera switched off
Powersupply removed
Camera switched on
USB-Cable connected – menu shows up
Select USB connect for HDD -> “connectd”
Command prompt, run as admin: “Exception: Firmware update error: AC adapter required”
Start console ===========================
[7936] Failed to execute script 'pmca-console' due to unhandled exception!
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell -f hdr-xr550v.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Traceback (most recent call last):
  File "pmca-console.py", line 100, in <module>
  File "pmca-console.py", line 84, in main
  File "pmca\commands\usb.py", line 377, in updaterShellCommand
  File "pmca\commands\usb.py", line 390, in firmwareUpdateCommandInternal
  File "pmca\usb\sony.py", line 557, in checkGuard
  File "pmca\usb\sony.py", line 531, in _sendWriteCommands
Exception: Firmware update error: AC adapter required
[6688] Failed to execute script 'pmca-console' due to unhandled exception!
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End console =============================

ma1co, 2021-08-12:
Ok, I need to think about how to proceed. I will let you know.

Thank you very much for your help so far, we are already a lot closer to supporting these cameras!

gkln, 2021-08-12:
I am in standby.
 
According to one of your previous mails, Do 2021-08-12 21:30
 
*	Download  <https://github.com/ma1co/OpenMemories-Tweak/files/6977999/HDR-XR550V.txt> HDR-XR550V.txt and place it in the same directory as pmca-console.
*	Re-connect your camera and run pmca-console updatershell -f HDR-XR550V.txt. You might have to press a button to restart the camera to updater mode.
*	You should now be greeted by another shell.
*	Run pull /dev/nflasha
*	Run bootloader
*	Run bootrom
*	Run exit to restart the camera
Please note that the camera may be initialized to factory settings during this process.
We should now have the different firmware files: bootrom, boot0 and nflasha.  <===
 
I am not sure about what this time should happen: already an update of the cameras firmware or the production of 3 more files, bootrom, boot0 and nflasha respectively.
 
However, I would appreciate if after this procedure the cam is still working or can be set back easily to an operable condition. ;)
 
cu

ma1co, 2021-08-13:
Oh, here we go:
0x102: The battery is attached to your camera. Remove the battery.
So Sony wants the AC adapter, but not the battery.

We are not trying to update the firmware here. We just need the firmware update mode to execute code. This won't touch the installed firmware. The only thing that could happen is that Sony decides to wipe all your settings (perform a factory reset).

gkln, 2021-08-13:
Steps:
Camera in condition shut down and disconnected from USB and power.
Battery removed
Power cord / AC adapter connected.
Display opened (turns camera on and allows access to USB socket)
USB connected
USB connect for HDD on display pressed (Win gives connect sound, Camera change display to “connected”)
Win pops up “automatic play(?)” for drive G. From several options “Import media” by PlayMemories home is preselected. Without selecting this Win reports in the system tray that GPS data were updated automatically and succesfully.
“Auto play” pop up closed.
Command prompt opened as admin.
changed to E:\Sw\HW_Treiber\Camera\SonyTweak where your SW resides.
è Windows Security Essentials pops up with msg 
è Trojan:Win32/Wacatac.B!ml quarantined, severe 2021-08-13 12:29
è Kategory: Trojan
è Description: This program is dangerous. It executes command of an attacker
è Suggested action: Remove this programm immediately
è  Elements: file:E:\Sw\HW_Treiber\Camera\SonyTweak\pmca-console-v0.17-63-g77b2991-win.exe
è  Online further information:
 <https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fWacatac.B!ml&threatid=2147735505&enterprise=0> Trojan:Win32/Wacatac.B!ml threat description - Microsoft Security Intelligence
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fWacatac.B!ml <https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fWacatac.B!ml&threatid=2147735505&enterprise=0> &threatid=2147735505&enterprise=0
 
At the moment your program is quarantined.
Please explain.

ma1co, 2021-08-13:
It happens from time to time that pmca-console / pmca-gui is wrongly detected as a virus. We had that in the past, and it seems to happen to most applications packaged with PyInstaller. It is clearly a false positive, the source code is open and the binary is built directly on our CI provider AppVeyor, so there is no way there could be a virus in there. I have submitted a false positive report to Microsoft and I hope that they will fix this as soon as possible.

gkln, 2021-08-13:
Ø  It is clearly a false positive, the source code is open and the binary is built directly on our CI provider AppVeyor
 
Well, this is not really convincing for me ;) On my behalf I am not the man to check source code neither can I delegate this job to anybody nor do I know AppVeyor, and even if this is a reliable organization / company, no one can know if this is true for all employees there who could play their own games. 
 <https://en.wikipedia.org/wiki/AppVeyor> AppVeyor - Wikipedia
 
But I trust you :) 
 
So, what now?
pmca-console-v0.16-10-gf546837-win.exe (I had still in the tweak-folder) was not marked as a trojan.
pmca-console-v0.17-63-g77b2991-win.exe was detected and somehow hidden away by windows. Until now I could not locate the “quarantine” folder to where MS Security Essential puts the files, even not with a search in internet. Those folders mentioned are all empty (windows explorer set to show also hidden files), also a “dir /ah *ca-console*” on c:> and e:> failed. It looks that windows completely masked this file.
I could try to recover the pmca-console within MSSE and allow the usage; alternatively V016 is still there and maybe would also work? Your opinion?

ma1co, 2021-08-13:
You can try v0.16, I think it should work.

gkln, 2021-08-13:
Steps:
Camera in condition shut down and disconnected from USB and power.
Battery removed
Power cord / AC adapter connected.
Display opened (turns camera on and allows access to USB socket)
USB connected
USB connect for HDD on display pressed (Win gives connect sound, Camera change display to “connected”)
Win pops up “automatic play(?)” for drive G. From several options “Import media” by PlayMemories home is preselected. Without selecting this Win reports in the system tray that GPS data were updated automatically and succesfully.
“Auto play” pop up closed.
Command prompt opened as admin.
changed to E:\Sw\HW_Treiber\Camera\SonyTweak where your SW resides.
run “pmca-console-v0.16-10-gf546837-win.exe updatershell -f HDR-XR550V.txt”
Start console ======================================================
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell -f HDR-XR550V.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Switching to updater mode
 
Waiting for camera to switch...
Please follow the instructions on the camera screen.                                                           [ *)
Operation timed out. Please run this command again when your camera has connecte
d.
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End console ========================================================
 
*) on the camera screen is displayed: “Follow the instructions on the computer”. (This is at the moment the end-condition; no more options to press or how to leave this condition; I let the cam so until your reply)
 
So PC and Cam reference each other
There are a couple of buttons on the cam – but no hint which I should press, maybe even more than one at a time.

ma1co, 2021-08-13:
The most similar camera that I could find which has a firmware update is the NX5. The [instructions](https://download.pro.sony/FNGP/protein/1237489124325/1237489126831.pdf) say that you should push the RESET button to switch modes. On the XR550V, this is the recessed button under the LCD that you have to push with a needle, just next to the POWER button. This will definitely clear all settings, but it might also switch to updater mode.

gkln, 2021-08-13:
Pushed „Reset“, cam display gets dark and stays so.
 
On command prompt:
Start console ==================
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell -f HDR-XR550V.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in updater mode
 
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...
 
Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
> 
>pull /dev/nflasha
Writing to .\nflasha...
Error: (2, 'DeviceIoControl', 'Das System kann die angegebene Datei nicht finden
.')                                                        [but there is one created with 28 MB]
>bootloader
Error: (2, 'CreateFile', 'Das System kann die angegebene Datei nicht finden.')
>bootrom
Error: Unknown command
>exit
Error: (2, 'CreateFile', 'Das System kann die angegebene Datei nicht finden.')
> 
End Console output ==========================================

ma1co, 2021-08-13:
Ok, we really seem to hit every possible problem... You are doing everything right, but it seems that the USB connection is a bit unreliable. Can you try the same procedure a few more times? You can also run the `bootloader` and `bootrom` commands without running `pull` first. Does the camera start up normally if you disconnect and reconnect power?

gkln, 2021-08-13:
Sent 20210813_183600 HDR-XR550V nflasha.zip in the meantime …
 
Steps undertaken:
USB unplugged
Resetbutton pressed
Cam recovers. Area and time set.
USB plugged. “Connected” (Battery still unmounted, running on AC)
 
Start console output ==================================
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell -f HDR-XR550V.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Switching to updater mode
 
Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...
 
Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bootloader
Writing to .\boot1...
>bootrom
Error: Unknown command
> 
End console output ===================================
 
20210813_183600 HDR-XR550V boot1zip uploaed
 
repeated the procedure, this time only bootrom
Start console output ==================================
 
a boot1 – file with 128 kb was produced, I zipped it and uploaded it as 
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell -f HDR-XR550V.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Switching to updater mode
 
Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...
 
Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bootrom
Error: Unknown command
> 
End console output ===================================

ma1co, 2021-08-13:
Oh, the bootrom command is not supported in v0.16, that's why it fails. The boot1 file looks good, thank you for uploading. Could you give nflasha another try? It should be about 128 MiB. If it fails again, you can also try to dump nflasha1, nflasha2, nflasha3, nflasha4, nflasha5, nflasha6 and nflasha7 seperately.

Thank you very much for your help, we are almost there!

gkln, 2021-08-13:
The 20210813_183600 HDR-XR550V nflasha.zip –content was not OK? app. 30 MB file inside.
 
I will try once more.
BTW: “exit” does not work, also ctrl-c, quit, qq fails. I have to close the command prompt and restart from the beginning.

gkln, 2021-08-13:
I changed the USB-Port on the PC.
 
Start console ======================
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.16-10-gf546837-win.exe updater
shell -f HDR-XR550V.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Switching to updater mode
 
Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...
 
Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>pull /dev/nflasha
Writing to .\nflasha...
>bootloader
Writing to .\boot1...
>bootrom
Error: Unknown command
>exit
Error: Unknown error
> 
End console =======================
 
uploaded 20210813_183600 HDR-XR550V nflasha 2.zip

ma1co, 2021-08-14:
Thank you, the nflasha file is now complete.

The exit command is broken in v0.16, it should work in the newest development releases.

The only thing missing now is the bootrom. Can you try if the [previous release](https://ci.appveyor.com/project/ma1co/sony-pmca-re/builds/40324460/artifacts) of pmca-console works for this?

We are almost done, thank you very much for your help!

ma1co, 2021-08-14:
Btw, Microsoft writes regarding the false virus positive: "We have removed the detection."

gkln, 2021-08-14:
I took the risk and recovered the 0.17-version from MSSE. No risk, no fun ;)
 
... and no battery
Start console output ================================
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell -f hdr-xr550v.txt
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Initializing firmware update
Switching to updater mode
 
Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
100%
Starting updater shell...
 
Welcome to USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bootrom
Writing to .\bootrom...
>exit
Cannot sync backup
Done
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End console output =================================
 
20210814_115000 HDR-XR550V bootrom.zip 6 kB uploaded.
 
BTW: I have access also to a Sony DSC-RX100M7. Interested?

ma1co, 2021-08-14:
Thank you for the bootrom dump. Interestingly, the bootrom is larger than expected and we have only dumped the first part of it. Can you try the same procedure again, but with this firmware file? [HDR-XR550V-2.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6987122/HDR-XR550V-2.txt) This should dump 16Kib instead of only 8Kib.

gkln, 2021-08-15:
20210815_113200 HDR-XR550V bootrom-1 uploaded.

ma1co, 2021-08-15:
Perfect, thank you very much. The firmware dump is now complete. Next, I will check in everything, so that the camera is supported in fwtool and pmca-console / pmca-gui without the additional firmware file. Thank you again very much for your help!

gkln, 2021-08-15:
I thank you for your efforts to improve usability of Sony camaras.
 
BTW: I got no yet an answer if you are interested in firmware dumps from Sony DSC-RX100M7

ma1co, 2021-08-18:
If possible, could you run two more commands on the HDR-XR550V updatershell?
`shell memtool dump 0x7a460000 4`
`shell cat /proc/cpuinfo`
This will make it easier to add it to qemu.

Regarding the DSC-RX100M7:
Yes, I would be interested in a dump. You should be able to dump the firmware by executing the following commands in serviceshell:
`pull /dev/nflasha`
`pull /dev/nflashaB0`
`pull /dev/nflashaB1`
Note that nflasha is about 2 GB, so it will take a while to dump.

Thank you again for your help!

gkln, 2021-08-18:
Sorry, need more distinct instructions.
Please provide exact commands I should run from the command-prompt (I think, admin rights are not necessary). It would save both our time. (Same for DSC-RX100M7: it is not clear for me, if I have first again to change to the lib-usb-drivers or if this would work immediately with the commands
“pmca-console-v0.17-63-g77b2991-win.exe updatershell pull /dev/nflasha”
 
 
Procedure so far as before, battery pack removed, USB connected in mass storage mode
.
Start console output ==================
E:\>cd E:\Sw\HW_Treiber\Camera\SonyTweak
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices
 
Querying mass storage device
Sony Camcorder is a camera in mass storage mode
 
Getting device info
Using firmware for model HDR-XR550V
 
Unknown device: HDR-XR550V
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell shell memtool dump 0x7a460000 4
usage: pmca-console-v0.17-63-g77b2991-win.exe [-h] [--version]
                                              {info,install,market,apk2spk,spk2a
pk,firmware,updatershell,serviceshell,guess_firmware,gps,stream,wifi}
                                              ...
pmca-console-v0.17-63-g77b2991-win.exe: error: unrecognized arguments: shell mem
tool dump 0x7a460000 4
 
E:\Sw\HW_Treiber\Camera\SonyTweak>pmca-console-v0.17-63-g77b2991-win.exe updater
shell memtool dump 0x7a460000 4
usage: pmca-console-v0.17-63-g77b2991-win.exe [-h] [--version]
                                              {info,install,market,apk2spk,spk2a
pk,firmware,updatershell,serviceshell,guess_firmware,gps,stream,wifi}
                                              ...
pmca-console-v0.17-63-g77b2991-win.exe: error: unrecognized arguments: memtool d
ump 0x7a460000 4
 
E:\Sw\HW_Treiber\Camera\SonyTweak>
End console output ==================

[4 more replies]