ma1co/OpenMemories-Tweak #267: [REQUEST] Star-eater fix

bcdub, 2018-03-11:
Could you add a way to disable the raw file noise reduction for bulb exposures? This would fix it for the a7r/s/II

Alternatively, just allowing longer timed exposures would fix it for the a7/r/s which would be great. Say 1, 2, and 4 minute exposures. 

Thank you for your time.

nikiosna, 2018-05-02:
I think there are a few other people who would like this also:
https://www.change.org/p/sony-remove-star-eater-en

mbarbeaux, 2021-05-11:
Any update on this enhancement yet ?
I suppose this must be hard as hell to find out how to disable it. Or maybe enabling longer exposure without using bulb ? Seems on the A7Smk1, this noise reduction well known as "star eater" by astrophotographers is only enabled on bulb exposure. If we could take e.g. 1-2-3-4-5 minutes long exposures without using bulb, that would be great !