ma1co/OpenMemories-Tweak #268: nex3n support

nex3n, 2018-03-16:
hi
i have nex3n and im willing to help with porting
where i should start?
looks like there is no firmware update exe to unpack
no android to dump it
maybe i didnt look deep enough
help

`Model:              NEX-3N
Product code:       0032825235
Serial number:      05152941
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model NEX-3N

Unknown device: NEX-3N`

ma1co, 2018-03-16:
You're probably the first person trying to run this on this particular model. This means that some additional steps are involved.

As a first step, we'll have to dump the firmware. We'll do this using the firmware update mode. First, we have brute force the model id of your camera. This won't install anything on your device. If you're interested to help, here's what you can do:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Download [0x31030000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/1820965/0x31030000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x31030000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

nex3n, 2018-03-16:
C:\Users\a\Desktop\a>pmca-console-v0.16-10-gf546837-win.exe guess_firmware -f 0x31030000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Model name: NEX-3N

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen2_0x31030022_0.dat

ma1co, 2018-03-17:
Thank you, the NEX-3N is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Connect the camera.
* Run `pmca-console updatershell`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

nex3n, 2018-03-17:
done
dont thank me so much i should thank you :)

ma1co, 2018-03-17:
Thank you very much for the firmware dump. Let me know if you have issues applying the tweaks.

closed 2018-03-17 by ma1co