ma1co/OpenMemories-Tweak #278: a7II can't switch to PAL

lucho9108, 2018-04-12:
I try the tweak and it doesn't work,
this is what I get:
Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera

then I hit install tweak and get this:
Starting task
Got task key
Cancelling plugin

help, please! I have a wedding tomorrow! and I need it in PAL as the first camera guy has it on PAL

ma1co, 2018-04-12:
Please try [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases/latest),

lucho9108, 2018-04-12:
I already did! This is what I get when trying pmca-gui
 Thanks for getting back to me but it’s not working, I am also trying on a
windows and doesn’t let me install it there

Luis Trujillo
Architect / Photographer
+61 452 365 847

ma1co, 2018-04-12:
No, this is the output of the browser plugin. This installation method is deprecated. Please use pmca-gui.

lucho9108, 2018-04-12:
But it's what I got from the program, not the plug in, do I have to
uninstall the plug-in? If so, how do I uninstall it.

Thanks

Luis Trujillo
Architect / Photographer
+61 0452 36 5847

lucho9108, 2018-04-12:
It's not with the plug in, have a look at this screenshot, I even quit
safari and Chrome! But still got this.

Thanks, Regards

Luis Trujillo
Architect / Photographer
+61 452 365 847

ma1co, 2018-04-12:
I cannot see the screenshot.

lucho9108, 2018-04-12:
Hi ma1co, I did it in another computer, maybe was because of the High
sierra mac software, I got it working, I installed it, but now I can't find
were to change it to PAL.
please if you can help me out where in the tweak app is it, or what steps
do I need to follow to get it on PAL. thank you again

regards,

lucho9108, 2018-04-12:
Hello, I did it, the thing is that I didn't turn it off so I didn't got the
NTSC/PAL selector, but now I do, thanks a lot! is there any tweak for
bracketing with only one press of the buttom with 2 second timer?

ma1co, 2018-04-13:
Currently, we don't understand enough of the firmware to implement such a feature.

closed 2018-04-13 by ma1co