ma1co/OpenMemories-Tweak #280: a6500 wont turn on, red light still blinking even in off position

aleberro, 2018-04-17:
Hi,

After installing the Open memories tweak and after setting the PAL/NTSC messague to be off i left the camera for a day or two. When in turned in on again after that couple of days the battery was at 3% so i inserted a new one. That was the last time the camera turned on. After that the camera doesnt turn on. Tried changing batteries, SD cards, everything... The camera is not recognized by the computer when plugged in so i cant update it or access the menu, Im running out of ideas by now! Please help!

ma1co, 2018-04-17:
How did you tweak the camera? Via the app or via the firmware update mode?
Hav you tried running the info command in pmca-gui when the camera is connected over USB?

aleberro, 2018-04-17:
Hi! 
I did it via the firmware update mode in a pc. 
I didn’t try the pmca-gui command, can you guide me on how to do it?

ma1co, 2018-04-17:
If you've done the tweak via the firmware update mode, you have either used pmca-console or pmca-gui. I want you to run either `pmca-console info` or click the "Get camera info" button in pmca-gui.

aleberro, 2018-04-17:
This is what the software tells me, it doesnt even recognize the camera,

ma1co, 2018-04-17:
What does the software tell you? I cannot see anything in your comment.

aleberro, 2018-04-17:
It says No devices found. Ensure your camera is connected.

It is as if the camera is locked with the read/write process. Is it
possible to  reset the camera via a  file in the SD card??

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

ma1co, 2018-04-17:
No, this is not possible as far as I know. Has your camera ever turned on normally after applying the tweak?

aleberro, 2018-04-17:
Yes it did, it worked perfectly. The latest change i had was changing from
pal to ntsc. When the camera restarted the message that warns you that its
running in NTSC appeared so i went to the app and disabled and re enabled
the "remove message" option. that was the last time it started up

ma1co, 2018-04-17:
In that case, I don't really think it is OpenMemories' fault (I can't know for sure though). Maybe your camera's hardware failed?

ma1co, 2018-04-17:
Sorry I cannot help. If you somehow manage to connect the camera to your computer, please let me know. I'm closing this for now.

closed 2018-04-17 by ma1co

aleberro, 2018-04-17:
Hey i managed to get to the SD card and found this:

INFO][MainActivity] application start
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: ILCE-6500
[INFO][InfoAdapter] Backup region: CX79700_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 4.1.2 (SDK 16)
[INFO][InfoAdapter] Java API version: 2.17
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .InfoActivity
[INFO][InfoAdapter] Model: ILCE-6500
[INFO][InfoAdapter] Backup region: CX79700_UC2
[INFO][InfoAdapter] Tweak app version: 0.11
[INFO][InfoAdapter] Android version: 4.1.2 (SDK 16)
[INFO][InfoAdapter] Java API version: 2.17
[INFO][onPause] .InfoActivity
[INFO][onResume] .VideoActivity
[INFO][onPause] .VideoActivity
[INFO][onResume] .RegionActivity
[INFO][onPause] .RegionActivity
[INFO][MainActivity] application end

Maybe it helps with something