ma1co/OpenMemories-Tweak #284: A6000 XAVC 60p issue - Possible workaround, assistance requested

Taylor0001, 2018-05-23:
Problem: When using OpenMemories Tweak and recording at 60p in XAVC-S, the recording consistently fails at 3:28:33 (tested with two separate A6000 cameras). Resulting in a corrupted, unrecoverable video file.

Possible Workaround: The code in VideoActivity.java sets the maximum time limit at 13H 1M 00S (at least in V 0.10, I don't understand how V 0.11 works). If this could be changed to 3H 28M 30S the video would stop recording three seconds before the failure point, preserving the recorded video.

I could use some help, if anyone has the knowledge and a few minutes it would be extremely helpful and appreciated if one could provide a custom APK version of OpenMemories Tweak with the shorter time limit.

I installed Android Studio and attempted to make the change myself but something just isn't working and I can't figure out how to compile the APK file.

mungewell, 2018-07-31:
As a work around; the time limit is "just" stored as values within the config database, these can be change individually either by:

1). Enable WiFi and Telnet with 'tweak', telnet to the device and use the 'bk.elf' command.
2). Using https://github.com/ma1co/Sony-PMCA-RE, which now supports the 'bk r and bk w' commands directly in updatershell.

Unfortunately I don't remember which keys need to be changed, and a quick search failed me...

ma1co, 2018-10-29:
Try this in updatershell:
```
bk w 003c0373 03
bk w 003c0374 1c
bk w 003c0375 1e
```
You can then check inside the tweak app if the limit was set correctly.

Taylor0001, 2018-10-30:
Thank you for the information. I have given that a try even though I have limited experience coding and using command prompt. Here's what I did:

In OpenMemories Tweak, I navigated to the Developer tab and enabled WiFi and Telnet.
I connected the camera to my home WiFi.
On my PC I enabled Telnet in Control Panel, I opened command prompt and entered the "telnet" command.
Then I entered the "o" command and connected to the camera using the IP displayed in the OpenMemories app.
Once connected I get this message:
```
BusyBox v1.13.4 (2012-07-17 15:11:04 JST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/ #
```
I attempted to enter the commands listed above and got this error message:
```
/ # bk w 003c0373 03
sh: bk: not found
```
I entered the "help" command and get this message:
```
Built-in commands:
-------------------
        . : [ alias bg break cd chdir continue echo eval exec exit export
        false fg hash help jobs kill let local pwd read readonly return
        set shift source test times trap true type ulimit umask unalias
        unset wait
```
I tried a few of the commands listed, the only one I tried that didn't throw an error was "set"
I entered the three commands as:
```
set bk w 003c0373 03
set bk w 003c0374 1c
set bk w 003c0375 1e
```
However, the time limit was still displayed in OpenMemories Tweak as 13h 01m 00s. I haven't yet tested the camera to see if it recording limit changed even though the displayed limit hasn't. I'm assuming if I did everything correctly the displayed limit would change. Any further guidance is greatly appreciated.

ma1co, 2018-10-30:
My instructions were for updatershell (run `pmca-console updatershell` while your camera is connected over USB). If you want to apply the changes over telnet, replace `bk` by `bk.elf` in my instructions.

Taylor0001, 2018-11-02:
Thank you very much. I was able to make the change via telnet to both of my A6000 cameras. I have tested one of them so far and the recording stopped at 3:28:30 with no issues, maintaining a fully working video file.

**For anyone else who may want to alter the maximum recording time here is a step by step guide based on my experience:**

Notes: 003c0373 = Hours, 003c0374 = Minutes, 003c0375 = Seconds. The two characters that follow are Hexadecimal numbers: 03 = 3, 1c = 28, 1e = 30. A decimal to hex converter can be used to find the hex equivalent for any time limit you want under the 13:15:21 theoretical limit described in the source code comments.

There are two ways to alter the maximum recording time; you can connect the camera to a wifi network using telnet or you can connect your camera via USB and use the updatershell function within the command line version of pcma, found here: https://github.com/ma1co/Sony-PMCA-RE/releases/tag/v0.17
If using the updatershell method the commands do not contain ".elf" and are as described by ma1co three comments above.

I used the telnet method, steps are as follows (Windows 10):

Prior to starting this procedure it is recommended you change the following setting on your camera:
Menu > Setup (toolbox icon) > submenu #2 > Pwr Save Start Time > Select: 30 Min
If not done, you may lose the connection after the camera screen dims.

1) On your PC open Control Panel
2) Find "Programs and Features"
3) Select "Turn Windows features on or off" on the top left side of the window
4) Scroll down to "Telnet Client," select the check box, and click OK (you may have to wait a minute while telnet is enabled)
5) Open Command Prompt
6) Type the command "telnet"
7) Turn on you camera
8) Navigate to the OpenMemories: Tweak app
9) Scroll right to the "Developer" tab
10) Select and turn on "Enable Wifi" and "Enable Telnet"
11) Scroll down to "Wifi settings"
12) Select "Access Point Set."
13) Select your home network (same network your PC is connected to)
14) Input password if required (press center button in scroll wheel to activate virtual keyboard), and select "OK"
15) In Command Prompt enter the command "o" (lowercase letter O)
16) In the Developer tab of OpenMemories: Tweak, under "Enable Wifi" there should be an IP address visible, enter the IP address into Command Prompt and press Enter
17) Command Prompt should now be connected to the camera.
18) Enter the each of the following three commands separately (press enter after each command is typed):
```
bk.elf w 003c0373 03
bk.elf w 003c0374 1c
bk.elf w 003c0375 1e
```
19) In OpenMemories: Tweak, navigate to the "Video" tab and check that the time limit under "Disable video recording limit" has changed. The check box will no longer be checked. If you attempt to check the box it will reset the limit to 13:01:00 and subsequently unchecking the box will set it back to 29:50. You will have to repeat the process if you check the box. If left untouched, the limit you set via telnet will be retained even if you turn the camera off, remove the battery, or reformat you SD card.
20) Once completed, you can close Command Prompt and disable telnet on you PC as well as turn off "Enable Wifi" and "Enable Telnet" in the Developer tab of OpenMemories: Tweak.