ma1co/OpenMemories-Tweak #289: Does installing tweak void my camera's warranty?

xyzyatna, 2018-07-06:
Just like the title:
Does installing tweak void my camera's warranty?

I want to remove the "running on ntsc" on startup, but I dont want my camera's warranty is void.

*Sorry for my bad english

ma1co, 2018-10-29:
Short answer: Nobody knows.

closed 2018-10-29 by ma1co

Peter-006, 2019-07-28:
> Just like the title:
> Does installing tweak void my camera's warranty?
> 
> I want to remove the "running on ntsc" on startup, but I dont want my camera's warranty is void.
> 
> *Sorry for my bad english

This tweak does not void your warranty (i think) because at difference of the Magic lantern, openmemories enable functions arleady existing

Srry 4 my bad english but i am italian