ma1co/OpenMemories-Tweak #300: Version 2.01 (Sony a6300)

matthewcosta, 2018-10-06:
Hey everyone,

I tried to install OpenMemories-Tweak. But my version on my Sony a6300 is v2.01.

The problem is, it connects, but i get an Error: usb.core.USBError: (Errno 13) Access denied (insufficient per missions). 

I hope you can solve or have a answer.

Thanks...
[image]

JMY1000, 2018-10-19:
I didn't have this issue with my a6300, but I did on my a7R. Switching the USB LUN Setting (the toolbox -> 4) solved the issue for me.

ma1co, 2018-10-29:
Please see my comment in [this issue](https://github.com/ma1co/Sony-PMCA-RE/issues/41).

closed 2018-10-29 by ma1co