ma1co/OpenMemories-Tweak #301: HDR CX 405 pmca.usb.InvalidCommandException

Ghorbel37, 2018-10-15:
I tried using openmemories with my cx 405 but it says pmca.usb.InvalidCommandException: Mass storage error: Sense 0x5 0x20 0x0. I know its not in the supported devices list but I thought it will work because its in the same family as cx440 (very similar software only wifi is missing and same release date). Also I tried downloading 0x11010000.zip and using the "pmca-console guess_firmware -f 0x11010000.zip" command but it didnt find a match. I really want to use 60p since I bought the camera in europe and I want to try the timelapse app.
many thanks

ma1co, 2018-10-15:
I'm not sure if unlocking 60p and running apps is possible on this camera. But as a first step, we can try to dump the firmware. If you're willing to help, please run the following command with [this](https://github.com/ma1co/OpenMemories-Tweak/files/2480518/0x51010000.zip) zip file and post the output: `pmca-console guess_firmware -f 0x51010000.zip`

Ghorbel37, 2018-10-15:
[image]
thanks for responding fast :)

ma1co, 2018-10-15:
Interesting, this is the same id as the PJ410... The next step is to dump the firmware. Here's how to do that:

* Connect the camera.
* Run `pmca-console updatershell -m HDR-PJ410`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot1` to `boot5` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

Ghorbel37, 2018-10-16:
I tried this but the console says PJ410 is an unknown camera. I tried both in mtp mode and mass storage.
[image]
[image]

ma1co, 2018-10-16:
Please use the development version of pmca-console available [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).

Ghorbel37, 2018-10-16:
I tried using the development version of pmca-console but it gives me a different error.
[image]
And I tried using the dev console yesterday and I got the same error so I tried downloading different versions of python and pywin32 and the problem didnt get resolved.
I dont know if there's a specific version I need or if its not related but I even tried it on a different pc and got the same error.
Thanks for the help you've been providing

ma1co, 2018-10-16:
Have you tried the binaries available [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts)?

Ghorbel37, 2018-10-16:
Yes

ma1co, 2018-10-16:
My bad. The issue should be fixed in the updated binaries on AppVeyor. Please try again :)

Ghorbel37, 2018-10-16:
Its okay, im executing the commands now and its working fine :)

Ghorbel37, 2018-10-16:
Done, that it took a long time to upload, I have a bad internet

ma1co, 2018-10-16:
Thanks, the CX405 is now supported in pmca-console / pmca-gui.

Ghorbel37, 2018-10-16:
Thanks you and keep up.
But one final thing I tried Unlock protected settings and it didnt work and when I try to unlock languages it says I need to unlock protected settings.

byChillout, 2018-12-11:
Same problem here :c 

i readed the sony cx405 is an android software based model but i cant install any aplication with the pmca, any idea about the reason?

Thanks u so much!

Serpentor, 2019-09-11:
I get the same errors on lines 123 and 77 with my FDR-AXP35.  Was this ever resolved?

Whatnoww, 2020-01-27:
> Thanks you and keep up.
> But one final thing I tried Unlock protected settings and it didnt work and when I try to unlock languages it says I need to unlock protected settings.
> In console it says unknown error and in gui it gives me this
> 
> Traceback (most recent call last):
> File "C:\projects\sony-pmca-re\pmca-gui.py", line 168, in do
> self.ui.shell.setTweakEnabled(self.id, arg)
> File "C:\projects\sony-pmca-re\pmca\usb\usbshell__init__.py", line 123, in setTweakEnabled
> self.USB_RESULT_ERROR_PROTECTION: 'Protection enabled. Please disable protection first.',
> File "C:\projects\sony-pmca-re\pmca\usb\usbshell__init__.py", line 77, in _req
> raise UsbShellException(errorStrings.get(r.result - 0x100000000, 'Unknown error'))
> pmca.usb.usbshell.UsbShellException: Unknown error

Do we know what this means yet? Additionally, I'm unable to install any apps from the GUI on the same CX405 even though it is supported.