ma1co/OpenMemories-Tweak #302: RX100 M5A MTP error 0x2006 occurs when try to install apk

nyk95srs, 2018-10-23:
Hello.
My RX100 M5A comes out with an error when I try to install apk on final step.
Here's the full info.
Is there any way to bypass this error?
I would really appreciate if you let me know
[image]

ma1co, 2018-10-29:
The RX100M5 is currently not supported. The android subsystem was removed, so there is no way to run custom apps anymore. It also uses new firmware encryption keys. Until these are recovered, nothing can be done.

closed 2018-10-29 by ma1co

nyk95srs, 2018-10-30:
@ma1co It's actually rx100 m5a, the new model released with M6
So is there no plan to support this model and M6 since the aos subsystem has been removed after M5?
Thanx for the kind answer! If there is any other way that I can try, could you plz let me know about it?

ma1co, 2018-10-30:
My bad, I meant that the RX100M5A is not supported. The old RX100M5 is supported.

For now, there is no way inside the new cameras, as Sony has changed the firmware encryption keys. Only if someone manages to recover them, we can see what we can do. But for now, there is no active work towards this.

nyk95srs, 2018-10-31:
@ma1co 
Is there anything I can help you with this for new M5A?

ma1co, 2018-10-31:
Right now, there isn't, but thanks for your offer.

superbroken, 2019-12-04:
i have Sony HX30V i got the same result is there any way?

ma1co, 2020-01-08:
@superbroken We can add the HX30V, but there is some work we have to do first. As a first step, we'd have to dump the firmware. We'll do this using the firmware update mode. First, we have brute force the model id of your camera. This won't install anything on your device. If you're interested to help, here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). If you're on os x, you have to wait for the next release (sorry).
* Download [0x21020000.zip](https://github.com/ma1co/Sony-PMCA-RE/files/3199140/0x21020000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x21020000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

nyk95srs, 2020-01-08:
Do you still dont have plan to add m5a m6 or the latest m7 😢

Android용 Outlook<https://aka.ms/ghei36> 다운로드

ma1co, 2020-01-08:
@nyk95srs  I'm trying to reverse engineer the maintenance mode (senser mode). We'll see how this goes.