ma1co/OpenMemories-Tweak #306: 4k raw Burst photo -» video hack

puredirect, 2018-11-14:
Im just thinking about, a7 cameras are capable of 10fps burst raw photo.
Maybe if we can turn off the curtain and forced to read out 24-30fps raw, it would be a fantastic :)
(we need very fast sd card but in post we can make videos from pictures)
They say video chip is not capable higher than 8 bit, but for photo it is 14bit.
It was a working solution on canon 5d cameras with magic lantern

Altirix, 2019-01-07:
will only work on the A7 III, but that doesn't have playmemories, and has 4k video already. so kinda pointless.

A7 & A7 II only do 5fps due to USB 2 data bus

Only A7S I / II, A7R II / III & A7III do 4k video

puredirect, 2019-01-07:
Please dont comment if you dont know what i would like!
I wrote 4K raw, most of these cameras do 4k compressed video...
Canon had this feature hacked in with magic lantern 10 years before, so you say it is impossible??
It was only hd resolution but sony cameras have much better chips.
Burst photo limit doesnt mean it is a phisical issue, it is maybe a datarate issue but if you read out 10 or 12 bit it is way less than 14 bit so if the camera can do 5-6 fps 14 bit data it can easly handle 25-30fps in 10 bit. In theory...

Altirix, 2019-01-11:
It seems you don't understand a few things, RAW can still be compressed without data loss (lossless compression, you get the same data back), otherwise, you would need 500MB/s to write 4k 30fps.

And again only A7S I / II, A7R II / III & A7III do 4k video of which only the A7S1/2 and the A7R2 support this hack.

You might be able to unlock 4k video on the A7 & A7II but no way in hell will you ever get RAW 4k

> Canon had this feature hacked in with magic lantern 10 years before. 

The Canon 5d III released in 2012 and the hack for 4k was introduced in 2017. Short 10 years.

If you bothered to read what i put. All of the A7 line up that this hack works on still use USB2 for SD card writes. This means you have a max write speed of about 35MB/s

The canon 5d III had a slot for CF cards. CF does upto 160MB/s, does it make a bit of sense as to why they can hack in 4k video and RAW video. The image processor in the 5d is obviously not powerful enough to compress it quickly enough of as effectively as the A7 but the CF card makes up for it.

10bit 30fps 4k sounds like it could work. RAW uncompressed that's 296MB/s (25fps is 247MB/s)

Bursting photos is 16bit @ 6fps 6k(24MP) is 278MB/s But of the A7 you can only burst for around 25 shots before the internal buffer fills up. id say this concludes the 35MB/s max is not enough to sustain this kind of fps after all that compression (the compression itself reduces the file size from 46MB to about 24MB).

I think its safe to say the data line from the sensor to the processor is fast enough to support 4k video however the data line from the processor to the storage is just so slow you would need processing power a magnitude greater than the A7 has to offer to compress it enough without compromising the quality.

you could very likely get 1080p RAW video, however, just not 4k.

abrock, 2021-04-07:
If you want a higher dynamic range you can use SLog2 (Picture profile 7). The video recorded by the camera will look weird and pale at first and you need sophisticated processing tools (e.g. kdenlive or cinelerra) to convert the footage to Rec709 and color-grade it. It's a complex topic so expect to invest a couple of hours/days to learn the camera settings and tools, but I found it gives nice-looking results.

Pegacy, 2022-06-05:
> I think its safe to say the data line from the sensor to the processor is fast enough to support 4k video however the data line from the processor to the storage is just so slow you would need [...]

That sounds good, is it possible to make 4k hdmi output for an external recorder?