ma1co/OpenMemories-Tweak #309: dsc-hx50v

vadcon78, 2018-11-22:
hi... i got a japanese locked version of the dsc hx50v.
i was hoping this tweak of your can be applied to the hx50v for the purpose of language change.
the camera i have is only displaying japanese and no other language option.
the pmca-gui won't work with it.
any other solution is welcomed.
you help is much appreciated.

ma1co, 2018-11-28:
The HX50V can probably be added. As a first step, we have to dump its firmware. If you're interested to help, here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). If you're on os x, you have to wait for the next release (sorry).
* Download [0x31020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/2624485/0x31020000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x31020000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

vadcon78, 2018-11-28:
okay... i got it.
i think.
here you go.

[image: image.png]

ma1co, 2018-11-28:
Sorry, I cannot see the image if you attach it to a mail. Please post it manually here: https://github.com/ma1co/OpenMemories-Tweak/issues/309

vadcon78, 2018-11-28:
it says

trying 100 firmware images
firmware update error : invalid version
success: found matching file gen2_0x31020051_0.dat

ma1co, 2018-11-28:
Thank you, the HX50 is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest build of pmca-console.
* Connect the camera.
* Run `pmca-console updatershell`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot1` to `boot5` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

vadcon78, 2018-11-28:
uploading now

ma1co, 2018-11-29:
Thanks for the dump! You can now try to unlock the languages using the Tweak tab in pmca-gui or the tweak command in `pmca-console updatershell`.

vadcon78, 2018-11-30:
hi...
thanks for the help so far.

this is what i get when i use pmcs-gui when using tweak tab

*Using drivers Windows-MSC, Windows-MTP*
*Looking for Sony devices*

*Querying mass storage device*
*Sony DSC is a camera in mass storage mode*

*Getting device info*
*Using firmware for model DSC-HX50V*

*Unknown device: DSC-HX50V*

anything else i need to do first before i click start tweaking?

this is camera info in case it is useful for anything

*Model:              DSC-HX50VProduct code:       0080813200Serial number:
    00039621Firmware version:   1.00GPS Data:           2018-11-30 00:00:00
- 2018-12-30 00:00:00*

vadcon78, 2018-11-30:
this is what i get when i use command prompt

*C:\Users\user\Downloads>pmca-console-v0.17-14-g886caa1-win updatershell*
*Using drivers Windows-MSC, Windows-MTP*
*Looking for Sony devices*

*Querying mass storage device*
*Sony DSC is a camera in mass storage mode*

*Getting device info*
*Using firmware for model DSC-HX50V*

*Initializing firmware update*
*Switching to updater mode*

*Waiting for camera to switch...*
*Please follow the instructions on the camera screen.*
*Initializing firmware update*
*Writing firmware*
*0%*
*100%*
*Starting updater shell...*

*Welcome to the USB debug shell.*
*Type `help` for the list of supported commands.*
*Type `exit` to quit.*
*>help*
*List of supported commands:*
*help                     Print this help message*
*info                     Print device info*
*tweak                    Tweak device settings*
*shell                    Start an interactive shell*
*shell <COMMAND>          Execute the specified command*
*push <LOCAL> <REMOTE>    Copy the specified file from the computer to the
device*
*pull <REMOTE> [<LOCAL>]  Copy the specified file from the device to the
computer*
*bootloader [<OUTDIR>]    Dump the boot loader*
*bk r <ID>                Read backup property*
*bk w <ID> <DATA>         Write backup property*
*bk s                     Sync backup data to disk*
*install <APKFILE>        Install the specified android app*
*exit                     Exit*
*>tweak*
*1: [ ] Disable video recording limit*
*       0h 29m 50s*

*2: [ ] Unlock all languages*
*       1 / 35 languages activated*

*3: [ ] Enable PAL / NTSC selector & warning*
*       Disabled*

*4: [ ] Enable USB app installer*
*       Disabled*

*5: [ ] Unlock protected settings*
*       Protection enabled*

*Enter number of tweak to toggle (0 to finish): 2*
*Error: Protection enabled. Please disable protection first.*

rgds

vadcon78, 2018-11-30:
okay.. i got it...
i went with option 5 to *Unlock protected settings.*

from there i'm able to unlock all the other tweak options.

the next step is just to figure out which of the japanese menu is the
language setting.

for the dsc-hx50v click MENU button, select the toolbox symbol, go to main
settings (Hammer symbol) - choose English.

thanks for the great help ma1co!

rgds
vadcon78

sionglow75, 2019-04-01:
> Thanks for the dump! You can now try to unlock the languages using the Tweak tab in pmca-gui or the tweak command in `pmca-console updatershell`.

Hi - thank you for the amazing work on this. I tried to change the 30 minute limit, and it shows [X] 
(for this option *1: [ ] Disable video recording limit*)

But when I do a recording it still turns off at 29 minutes. Any ideas? Thank you! @ma1co

sionglow75, 2019-04-01:
Ah never mind ... it works if I change this to AVCHD. Thank you again @ma1co

closed 2019-06-11 by ma1co

edsmart, 2020-10-29:
Hi ma1co I am trying to change a little Cybershot DSC-WX100, I did follow the comment above, run pmca-console guess_firmware -f 0x31020000.zip command but didn't have luck with.
This is what I got: 

C:\Users\ebuitrag>pmca-console guess_firmware -f 0x31020000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Model name: DSC-WX100

Trying 100 firmware images
Fail: No matching file found

Is there anything I can do ?

Many thanks

ma1co, 2020-10-29:
Can you try with this file instead? [0x21020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5460460/0x21020000.zip)

edsmart, 2020-10-30:
Thank you ! 

C:\Users\ebuitrag>pmca-console-v0.17-35-gb504119-win guess_firmware -f 0x21020000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Model name: DSC-WX100

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen2_0x21020010_0.dat

C:\Users\ebuitrag>pmca-console-v0.17-35-gb504119-win updatershell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-WX100

Unknown device: DSC-WX100

I found the match but again I got unknown device 

Thank you

ma1co, 2020-10-30:
Thank you, the WX100 is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest build of pmca-console.
* Connect the camera.
* Run `pmca-console updatershell`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

edsmart, 2020-10-30:
yes, i tried different ways, but i'm not sure how's the right syntax?

C:\Users\ebuitrag>pmca-console-v0.17-35-gb504119-win updatershell pull /dev/nflasha
usage: pmca-console-v0.17-35-gb504119-win [-h] [--version]
                                          {info,install,market,apk2spk,spk2apk,firmware,updatershell,guess_firmware,gps,stream,wifi}
                                          ...
pmca-console-v0.17-35-gb504119-win: error: unrecognized arguments: pull /dev/nflasha

can you help me with that ?

ma1co, 2020-10-30:
Just run the following command: `pmca-console-v0.17-35-gb504119-win updatershell`
It will guide you through switching to updater mode. Once your camera has switched, it will show you a promt (`>`).
Once you are there, enter the following: `pull /dev/nflasha`

edsmart, 2020-10-30:
C:\Users\ebuitrag>pmca-console-v0.17-35-gb504119-win updatershell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-WX100

Unknown device: DSC-WX100

C:\Users\ebuitrag>

ma1co, 2020-10-30:
Download [pmca-console-v0.17-36-g12d8b13-win.exe](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) first, then run `pmca-console-v0.17-36-g12d8b13-win updatershell`

edsmart, 2020-10-30:
uploaded

ma1co, 2020-10-30:
Thanks for the dump! The WX100 should now be supported in pmca-gui.

edsmart, 2020-10-30:
THANK YOU VERY MUCH !!!
you're awesome ! thanks a lot !!! 
have a happy day !!! 
thanks

damjep, 2021-05-25:
hi i need help with this dsc wx5 i want to install tweaks but couldnt 

and this is the message

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-WX5

Unknown device: DSC-WX5

ma1co, 2021-06-03:
The DSC-WX5 should be supported in the [development builds](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). There are no tweaks available, however, since this is a very old camera.

alexflear, 2021-07-19:
Hello ma1co!
Do you know if it works on Sony a7c ?? I live in Japan and want to buy the camera but the menu is all only in japanese :(

jojoanamarie, 2022-07-11:
Hello @ma1co I recently acquired a Japanese version of Sony HDR CX700. I'm trying to follow the discussion here in the forum to change the language in my camcorder, but I'm hitting a wall. When I try to install OpenMemories:Tweak this is what I get. 

> Querying mass storage device
> Sony Camcorder is a camera in mass storage mode
> 
> Model:              HDR-CX700V
> Product code:       0002091301
> Serial number:      00011519
> Firmware version:   1.00
> GPS Data:           2012-04-21 00:00:00 - 2012-05-21 00:00:00
> 
> Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
> Looking for Sony devices
> 
> Querying mass storage device
> Sony Camcorder is a camera in mass storage mode
> 
> Switching to app install mode
> Error: This device does not support apps

I hope you can help me figure this out. Thank you so much and I appreciate you doing this for us!

jojoanamarie, 2022-07-11:
Also, a quick note that I tried running the pmca-console-v0.18-win but it just opens for a second and then automatically closes.

jojoanamarie, 2022-07-11:
Update: I've figured it out :)

I just needed to connect the charger to the camera while it's connected to my laptop. I was able to use the GUI :D

brent-williams, 2022-10-09:
> Hello ma1co! Do you know if it works on Sony a7c ?? I live in Japan and want to buy the camera but the menu is all only in japanese :(

Hi @alexflear, long-shot but my buddy here in Australia is Japanese and purchased the A7C only to find it's locked to English (he'd prefer Japanese).  Again long-shot but we're traveling to Japan (Chiba) in January, maybe you'd like to buy locally and do a trade?

velobianchi, 2023-01-04:
hello ma1co, I also have a Japanese Cyber-shot DSC HX50v.
the pmca-gui won't work with it, 
I'm not sure if I'm doing this correctly,
should I install and run pmca-console v0.17 first then install run pmca-gui?
I ran pmca-gui but I get error Unknown device: DSC-HX50V

 I see that you fixed this issue with another user(same camera model), hoping that you can help me out too, thanks in advance.

velobianchi, 2023-01-05:
I installed newest GUI V.18

Querying MTP device
Sony Corporation DSC-HX50V is a camera in MTP mode

Model:              DSC-HX50V
Product code:       0080813200
Serial number:      00047825
Firmware version:   1.00
GPS Data:           2023-01-04 00:00:00 - 2023-02-03 00:00:00

Using drivers Windows-MSC, Windows-MTP, Windows-vendor-specific, libusb-MSC, libusb-MTP, libusb-vendor-specific
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-HX50V is a camera in MTP mode

Switching to app install mode
Error: This device does not support apps

- - - - - - - - - - - 
I installed pmca Console also, the window (shell) appears for 1 sec. then disappears.
- I used a command prompt to run pmca-console but the command is not recognized as an internal command by windows. 
- - guess _'m stuck for now. Wonder if you cvan help, thanks.
- _

velobianchi, 2023-01-05:
Microsoft Windows [Version 10.0.19044.2364]
(c) Microsoft Corporation. All rights reserved.

C:\Users\Manny>cd downloads

C:\Users\Manny\Downloads>pmca-console guess_firmware -f 0x31020000.zip
'pmca-console' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\Manny\Downloads>pmca-console updatershell
'pmca-console' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\Manny\Downloads>

velobianchi, 2023-01-05:
Success now! 3rd time's a charm.
I used the new V.178 of the GUI.exe, switch to the 'Tweaks' tab, click on 'Start Tweaking' button,
then it started Initializing firmware update.
I should mention to switch the USB to 'Mass Storage' setting on camera body.

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-HX50V

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
100%
Done
- - - - - 
- go back into camera settings, page 1/5, choose the 3rd line down, select button and choose English. That's it.  Thanks so much for all the work you put into this. Now I can thoroughly enjoy my new to me Cybershot DX50V. smile! :-)

haiau87, 2023-04-12:
> Thành công ngay bây giờ! Lần thứ 3 là một sự quyến rũ. Tôi đã sử dụng V.178 mới của GUI.exe, chuyển sang tab 'Tinh chỉnh', nhấp vào nút 'Bắt ​​đầu Tinh chỉnh', sau đó nó bắt đầu Khởi tạo cập nhật chương trình cơ sở. Tôi nên đề cập đến việc chuyển cài đặt USB sang 'Bộ nhớ chung' trên thân máy ảnh.
> i have sony hx30v , can you guide me, Thank You !
> Truy vấn thiết bị lưu trữ dung lượng lớn Sony DSC là máy ảnh ở chế độ lưu trữ dung lượng lớn
> 
> Nhận thông tin thiết bị Sử dụng chương trình cơ sở cho mẫu DSC-HX50V
> 
> Đang khởi tạo cập nhật chương trình cơ sở Chuyển sang chế độ cập nhật
> 
> Đang đợi máy ảnh chuyển đổi... Vui lòng làm theo hướng dẫn trên màn hình máy ảnh. Đang khởi tạo cập nhật firmware Ghi firmware 100% Done
> 
> * vào lại camera setting, trang 1/5, chọn dòng thứ 3 trở xuống, chọn button và chọn English. Đó là nó. Cảm ơn rất nhiều cho tất cả các công việc bạn đưa vào này. Bây giờ tôi có thể hoàn toàn tận hưởng chiếc Cybershot DX50V mới của mình. nụ cười! :-)

lavasnaill, 2023-12-11:
@ma1co hello!! so i just got a sony cyber shot DISC-HX5V and i know 0 things about coding, softwares, or anything computers (besides the obvious things) im trying to get the language to be english and stumbled upon this. I followed the original steps and all i get is a *no tweaks found* im very confused and could use your help or anyone’s matter of fact, thank you!!

lavasnaill, 2023-12-12:
I think it’s just because the model isn’t added to the tweak…

TGE3416, 2024-01-14:
> C:\Users\ebuitrag>pmca-console-v0.17-35-gb504119-win updatershell Using drivers Windows-MSC, Windows-MTP Looking for Sony devices
>
> Querying mass storage device Sony DSC is a camera in mass storage [...]

hey @edsmart @ma1co , do any you would mind explainingme  how you got the updatershell to work? I have been trying to change a DCS-WX50, but whenever I run the command I get the same issue you have here, and I dont understand what exactly I should be seeing instead to know if it worked or not. I would appreciate it if any of you could help me out

jcpancho, 2024-01-29:
> @ma1co hello!! so i just got a sony cyber shot DISC-HX5V and i know 0 things about coding, softwares, or anything computers (besides the obvious things) im trying to get the language to be english [...]

same issue i have

fiji0708, 2024-02-22:
@ma1co hii! do you mind explaining me how to change the language of my cybershot dsc-hx5v? i tried downloading the files given but it didn’t work 🥲 
it said that switching to app install mode 
error: this device does not support apps
hoping for your reply^^ ty!

sabrinakbui, 2024-07-16:
Hi! I was wondering if anyone has had any luck with this on a Cybershot DSC-T77? I don’t think I saw it on the list of supported cameras. Thank you so much!

cors27, 2025-07-12:
Is it possible to change the language of my Sony DSC-HX60V camera from Japanese to English? If so, how can I do it?