ma1co/OpenMemories-Tweak #312: Cant install applications anymore with USB connection to A7r II

teijount, 2018-11-29:
Referring to discussion:  A7RII Open Memories Tweak Black screen #214, I had same problem with my camera having already fw 4.00. I could fix it and and it has all tweaks on.  By initialising I lost all applications and I tried upload them by pmca-gui-v0.17-win and it was no anymore possible due communication error, how can I fix this? Applications from official Sony site with  Wifi I can download! 

Starting task
Starting communication
Uploading 0%
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 76, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 291, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 84, in installApp
  File "C:\projects\sony-pmca-re\pmca\installer\__init__.py", line 84, in install
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it.
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 116, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 346, in updaterShellCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 251, in getCameraInfo
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 240, in _sendCommand
AttributeError: 'SonyMtpAppInstaller' object has no attribute 'sendSonyExtCommand'

Arnie97, 2019-04-02:
I got the `ConnectionRefusedError` too, but with a different stack trace:
```
$ pmca-console-v0.17-win install
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Switching to app install mode
Waiting for camera to switch...
Starting task
Starting communication
Uploading 0%
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-console.py", line 96, in <module>
  File "C:\projects\sony-pmca-re\pmca-console.py", line 72, in main
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 291, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 84, in installApp
  File "C:\projects\sony-pmca-re\pmca\installer\__init__.py", line 84, in install
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
[4876] Failed to execute script pmca-console
```

```
$ pmca-console-v0.17-win info
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              ILCE-6000
Product code:       0032826900
Serial number:      03051197
Firmware version:   3.21
Lens:               Model 0x12a08019 (Firmware 1.04)
```

Arnie97, 2019-04-02:
Btw if anyone is having similar issues, please try the [the web-plugin based installer](https://sony-pmca.appspot.com/plugin/app/com.github.ma1co.openmemories.tweak).
I think it uses the same method as the Sony official site, hence just works (at least in my case, ymmv).