ma1co/OpenMemories-Tweak #322: Sony A6000 silent shutter and closing shutter after turning off

Mrowkarz, 2019-02-24:
Hi!
I wonder if it would be possible to turn on electronic shutter in A6000. It's only software or hardware limitation?
Also it would be great to make the shutter closing after turning off the camera. It could cover the sensor during changing lenses and keeping it away from dust. I think it is possible to do in software.

br3aksoul, 2021-01-04:
It would be nice to have silent shutter in Sony a6000. Even a low resolution image would be great. Maybe a 6mpx or even a full HD picture. I know I can grab a frame in post processing, but a 1080p silent picture mode is more usefull. Raw would be great, but jpg is ok. The readout speed should'nt be a problem with such low resolution. I know that chances are low, but hope dies last.

Hondune, 2022-04-11:
I have dumped the bin files for silent shooting on/off here https://github.com/ma1co/OpenMemories-Tweak/issues/397 but have been unable thus far to figure out which setting(s) need to be changed to get this working on other cameras. I can see references to silent shooting settings in the dumped firmware all the way down to the a5000 so i think it should be possible. Maybe someone else will have better luck