ma1co/OpenMemories-Tweak #323: DSC-RX1 not recognised

cookieboyadam, 2019-03-09:
Hi there,

I'm trying to figure out if I can somehow enable clean HDMI output on my RX1, I've tried running pmca-gui-v0.17-win but despite the camera being supported as of last year, I have a product code to the RX1's listed here. 

Model:              DSC-RX1
Product code:       0080812110
Serial number:      05526383
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-RX1

Unknown device: DSC-RX1

This is as far as I get when I hit start tweaking. The option I'm hoping to change is HDMI Info Display

fnicastri, 2019-06-14:
I have the same issue,
any news?

GitSak, 2020-05-26:
the same issue

ctflower, 2023-02-02:
curious if anyone found a fix to running this on the RX1