ma1co/OpenMemories-Tweak #332: My ILCE-6400 is in Chinese, no way to switch it to English?

Danhm87, 2019-04-26:
I just bought a Sony a6400 from an official Sony store on the Chinese marketplace, TaoBao. I live In China, and buy all my equipment from Taobao.

I was baffled to find that the camera only had Chinese, Korean, Hindi, and some other obscure languages available. No English? No problem, I'll just install the English firmware... I mean... the buttons on the camera are in English, surely it is just a software problem.

Wait... no firmware available on Sony's website??? what the heck???

I tried the OpenMemories: Tweak but apparently it doesn't work for the A6400... only the 6300 and the 6500.... FML

Store I bought it from will not return it.... Please, someone help me.... there has GOT to be a way.....

omiazad, 2019-05-07:
I think you have to wait till they add support for a6400

yonov-lelekov, 2019-07-03:
I have same problem. But i bought japanese camera)))