ma1co/OpenMemories-Tweak #335: a6500 1.06 firmware

effysh, 2019-05-30:
dose open memories tweak support the new 1.06 firmware ?

robertozuena, 2019-06-16:
follow

rpb1515, 2019-10-05:
Hello, I have the version of firm 1.06 on the a6500 and I have not installed it yet because I am reading that it seems that it is not going. Is this correct?
I would be interested to know:
1) Would it be possible to allow the record button to be set on the front shutter (shutter button) in the option that has the menu to customize these buttons?
2) There is possibility to make the screen not dim when it is set to 4k or 1080 100p / 120p. Thank you
3) Is it true that you write a log on the SD Card and that if it is changed or formatted does the unlocking work? or on the contrary it is internal load.
Thanks

gouletpalex, 2019-12-04:
Hi, I had Firmware 1.05 and I had installed OpenMemories Tweak. I was wondering the same thing as you (if updating to 1.06 would break the tweak from working). I could not find the answer, so I did the update anyway.

It turns out that after the update, everything still works just as before! From what I understand, the OpenMemories Tweak uses the API in order to work, so it seems like it's independent from the Firmware version.

Cheers!

slam51, 2019-12-18:
1.06 will prevent the tweak from installing but not break it.  if you have it is at 1.05, it will install and all settings will permanently be changed.  but 1.06 prevent the tweak from installing.

gouletpalex, 2019-12-20:
slam51: good to know!

Of course Sony would prevent their users to use the Tweaks with this new firmware... Not only are we not getting much new functions with this update, but apparently Sony doesn't want its users to be able to enable their own functions on the side... To say the least Sony is far behind Panasonic with functional firmware updates!

robertozuena, 2019-12-20:
> Ciao, avevo Firmware 1.05 e avevo installato OpenMemories Tweak. Mi chiedevo la stessa cosa di te (se l'aggiornamento a 1.06 avrebbe interrotto la modifica dal funzionamento). Non sono riuscito a trovare la risposta, quindi ho fatto comunque l'aggiornamento.
> 
> Si scopre che dopo l'aggiornamento, tutto funziona ancora come prima! Da quanto ho capito, OpenMemories Tweak utilizza l'API per funzionare, quindi sembra indipendente dalla versione del firmware.
> 
> Saluti!
grazie, allora adesso posso fare l'aggiornamento con serenità?

slam51, 2019-12-20:
The reason they cut that off is to prevent virus. I just wish they give me the ability to roll back to an earlier version.

ma1co, 2020-01-08:
I am not aware that firmware version 1.06 would break the tweaks. If this is the case, please open a separate issue with the exact error message.

closed 2020-01-08 by ma1co