ma1co/OpenMemories-Tweak #337: Support for SONY DSC HX9V?

udippel, 2019-06-06:
I tried my best, with Internet Explorer, and when this didn't work with the lastest installer, GUI, on W10, v0.17.
The camera is recognized, but at all efforts it failed (see attachment). I tried starting the program, then the camera, the camera followed by the program. I had power via USB enabled and disabled. 
I also tried the tweak, but the error message is almost identical.

udippel, 2019-06-06:
[image]

udippel, 2019-06-07:
I tried a tad harder today, with mass storage, and IE with the plugin. The latter always ends at 'network error'. Therefore I also tried the stand-alone GUI and command line versions. About the best was GUI with mass storage, because it recognized the camera and the firmware installed and a few more. However, installing the memory tweak failed. 
Next round was with the Zadig drivers, but that made everything worse. 

I'd be most happy to help out with whatever possible. Just ask.

Uwe

ma1co, 2019-06-24:
See https://github.com/ma1co/Sony-PMCA-RE/issues/152

closed 2019-06-24 by ma1co