ma1co/OpenMemories-Tweak #339: Sony rx1r (not rx1rm2) japan edition language change #261

damarogit, 2019-06-11:
Hi,

There was already an identical ticket [here](https://github.com/ma1co/OpenMemories-Tweak/issues/261) but it's already closed. 

I have a Sony RX1R, gift from Japan.
I want to know if it is possible to install english language on this camera. Now it's only Japanese. 
I wanted to try with Open Memories, but i didn't find my camera on the list with supported cameras by the app.
It appeared to be possible, i've done all the steps from the previous post, I even have nflasha and boot1 .

Thank you

ma1co, 2019-06-11:
Download [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases), go to the "tweaks" tab and click "start tweaking".

closed 2019-06-11 by ma1co

damarogit, 2019-06-12:
Thanks a lot, i will try that tonight after work.

damarogit, 2019-06-12:
Problem solved. 
Many thanks!
I'm really grateful!

fnicastri, 2019-06-14:
Hi, it is possible to do the same operation on a RX1 non R model?
I have an international version but it lacks the Italian language.

Thanks

halinh12sinh, 2020-02-18:
> Download [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases), go to the "tweaks" tab and click "start tweaking".

Do

> Problem solved.
> Many thanks!
> I'm really grateful!

Hi @damarogit ,
Can you please provide full manual. I have the same camera with you.
I tried: connect camera to pc( windows 10)-> Turn on camera->Open PMCA-gui->go to the "tweaks" tab and click "start tweaking"=> But nothing change.
Do I have to install any drivers? Or change any setup on my camera?

yulinwen, 2020-02-28:
The update process has failed.
It can recognize my camera model however it failed to update the firmware ("start tweaking" failed).

The screen shows:
Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              DSC-RX1R
Product code:       0080814200
Serial number:      06011312
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-RX1R

Unknown device: DSC-RX1R

yulinwen, 2020-02-28:
Ok!! Problem solved!!!
The thing is: to get the latest pcma 0.17 gui, not from the one provided in this thread!!!
Then bingo!!

Getting device info
Using firmware for model DSC-RX1R

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%

halinh12sinh, 2020-02-29:
Wow man! Can you pls share the software with me!Please

yulinwen, 2020-02-29:
Download the software here: "pmca-gui-v0.17-24-gb20a868-win.exe" https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts and follow the same instruction above. This beta version has support for RX1R.

halinh12sinh, 2020-02-29:
It's a amazing. How could say thank you!!!

yulinwen, 2020-02-29:
Thank the author ma1co! He's done the work. I was just sharing my experience.

GitSak, 2020-05-26:
Thank you so much !!!

jimmij1968, 2022-06-17:
hello guys this version pmca-gui-v0.18-win.exe 

100% will work with rx1r japanese to english? please confirm guys?

jimmij1968, 2022-06-17:
just confusing which one to use? pmca-gui-v0.18-win.exe or the pmca-console-v0.18-win.exe

jimmij1968, 2022-06-17:
> Download [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases), go to the "tweaks" tab and click "start tweaking".

> Download [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases), go to the "tweaks" tab and click "start tweaking".

hello, which start tweaking? start tweaking( updater mode) or the start tweaking (service mode) thank you if you reply

Ksuyoon, 2022-07-24:
> [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases) 를 다운로드 하고 "조정" 탭으로 이동하여 "조정 시작"을 클릭하십시오.

I even signed up for GitHub on purpose to write comments. You are truly a genius.

sunino, 2022-11-27:
Same here, I cant thank you enough ma1co, u r our hero!