ma1co/OpenMemories-Tweak #342: Alpha7 R2 firmware upgrade

abrock, 2019-06-28:
I have a a7r2 and the tweak works like a charm, I made several recordings of ~2h concerts without problems, thank you very much.

My camera has firmware version 4.00 installed and Sony recently released an firmware upgrade (4.01), did anyone try to

a) install the tweak on a camera with the new firmware or
b) upgrade a camera which has the tweak already?

MarkKrowd, 2021-03-04:
Hi, have the version 4.01 and it's working fine (haven't installed Tweak earlier), but I am experiencing other issues with this new firmware, so if you don't have problems with you camera now, I would recommend to not upgrade your firmware ;)

abrock, 2021-04-07:
Thank you very much for your comment @MarkKrowd. I didn't notice any problems with my camera so I'll just leave it.

closed 2021-04-07 by abrock