ma1co/OpenMemories-Tweak #343: Request: a5100 completely disable jpeg noise reduction

rzvz, 2019-07-04:
I was wondering if there is any way to completely disable noise reduction for jpegs? 
even with NR set to low, from iso 400 onwards i can clearly see the NR pattern, and i don't really want to shoot RAW just for that. to clarify I'm not talking about the long exposure NR. 
Has this maybe been done before?

MetroidFusion, 2021-01-30:
I'm looking for a similar solution. The A6000 has the option to turn 'High ISO NR' off. If we could somehow modify the A5100 firmware, as the two cameras are very similar, that could be a solution. However I don't have any experience with this. If somebody has, please let us know :)

mungewell, 2021-04-01:
@MetroidFusion Do you have access to the A6000?

Do a dump of the 'backup.bin' with and without the option enabled to see if 'we' can located the key that is used to hold this parameter. It may be that we can set the same key of the other cameras...