ma1co/OpenMemories-Tweak #346: Tweak not working on Sony Rx100M5A

Mantatad77, 2019-07-27:
Hi There 
Really cool app but could get it run on 
Model:DSC-RX100M5A
Product code:       0080817760
Firmware version:   1.00
Trying to get more than 5 minutes  4k video underwater :)

Tried the install app:
Using drivers Windows-MSC, Windows-MTP
Looking for Sony device
Querying mass storage device
Sony DSC is a camera in mass storage mode
Switching to app install mode
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 76, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 274, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 37, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 306, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 244, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 67, in sendSonyExtCommand
  File "C:\projects\sony-pmca-re\pmca\usb\__init__.py", line 36, in _checkResponse
pmca.usb.InvalidCommandException: Mass storage error: Sense 0x5 0x20 0x0

Tried the Tweaks:
Getting device info
Using firmware for model DSC-RX100M5A

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 116, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 360, in updaterShellCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 373, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 527, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 501, in _sendWriteCommands
Exception: Firmware update error: Invalid model

Many thanks

David

Peter-006, 2019-07-28:
Try to switch to MTP (becauz "Sony DSC is a camera in mass storage mode") or change usb cable

Mantatad77, 2019-07-28:
Hi Peter

Tried new cable and set MTP

Install app:

Querying MTP device
Sony Corporation DSC-RX100M5A is a camera in MTP mode

Switching to app install mode
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 76, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 274, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 37, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 306, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 244, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 98, in sendSonyExtCommand
  File "C:\projects\sony-pmca-re\pmca\usb\__init__.py", line 81, in _checkResponse
pmca.usb.InvalidCommandException: MTP error 0x2006

Tweaks:

Getting device info
Using firmware for model DSC-RX100M5A

Initializing firmware update
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 116, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 360, in updaterShellCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 373, in firmwareUpdateCommandInternal
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 527, in checkGuard
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 501, in _sendWriteCommands
Exception: Firmware update error: Invalid model

Regards

David

Peter-006, 2019-07-28:
Try to install the tweak from the site (of you have some tips for reduce overheating, tell them to me plz)(srry 4 my bad english)

Mantatad77, 2019-07-29:
Hi Peter

Tried again and got this issue

Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

With regards to overheating found this https://www.youtube.com/watch?v=rCYwQVtlSyY
and this one seems credible https://www.dpreview.com/forums/thread/3998838

Regards

David

Peter-006, 2019-08-02:
Try to download the apk from this site (  https://github.com/ma1co/OpenMemories-Tweak/releases  )
and install it using pmca.gui app

Mantatad77, 2019-08-03:
Hi Peter

Had a go and got the following message,Is there anything i can give you to help your trouble shoot like a copy of firmware ?

Regards

David

[https://avatars1.githubusercontent.com/u/12417797?s=400&v=4]<https://github.com/ma1co/OpenMemories-Tweak/releases>
Releases · ma1co/OpenMemories-Tweak · GitHub<https://github.com/ma1co/OpenMemories-Tweak/releases>
github.com
Unlock your Sony camera's settings. Contribute to ma1co/OpenMemories-Tweak development by creating an account on GitHub.

Peter-006, 2019-08-03:
I think tour camera Is uncompatible ( check for security  https://github.com/ma1co/OpenMemories-Framework/blob/master/docs/Cameras.md )