ma1co/OpenMemories-Tweak #354: HDR-PJ630v (japanese version) Language change.

m-hassan-memon, 2019-11-18:
Hey,
First of all thanks a lot, this helped me changing the language of my 3 sony cameras.
Now the issue is that I have many more Sony and other brand's Japanese version cameras and I want it to change it's language from Japanese to English.
Right now I want to change my **Sony HDR PJ-630** from Japanese to the English language.
Please help me with this, I believe that you can do this, else please refer me a good solution as I mentioned above that I have so many different brands cameras.
**### Once Again Thanks A Lot**

ma1co, 2020-01-08:
@m-hassan-memon  As a first step, we'd have to dump the firmware. We'll do this using the firmware update mode. First, we have brute force the model id of your camera. This won't install anything on your device. If you're interested to help, here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts). If you're on os x, you have to wait for the next release (sorry).
* Download [0x31010000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/4035171/0x31010000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x31010000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

BasitShaikh01, 2020-07-13:
Sir This pmca-console didn't work 
Please tell me how to change my Sony HDR PJ-630V from Japanese to the English language.

BasitShaikh01, 2021-01-02:
pmca-console not executing in window8.1.I want to change my Sony HDR PJ-790v from Japanese to the English language.
[problem.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5760474/problem.zip)

ma1co, 2021-01-03:
@BasitShaikh01 You have to run the program in the windows command line. Place the .exe file and the .zip file in the same directory, [open the command line](https://www.computerhope.com/issues/chusedos.htm), run `cd C:\path\to\the\directory` and then run `pmca-console-v0.17-39-gd646def-win.exe guess_firmware -f 0x31010000.zip`

BasitShaikh01, 2021-01-03:
@ma1co 
[problem.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5762166/problem.zip)

BasitShaikh01, 2021-01-03:
@ma1co 
[problem.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5762180/problem.zip)
[problem.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5762179/problem.zip)

BasitShaikh01, 2021-01-03:
@ma1co 

[problem_1.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5762181/problem_1.zip)
[problem.zip](https://github.com/ma1co/OpenMemories-Tweak/files/5762185/problem.zip)

m-hassan-memon, 2021-01-03:
> @m-hassan-memon As a first step, we'd have to dump the firmware. We'll do this using the firmware update mode. First, we have brute force the model id of your camera. This won't install anything on [...]

I do have CX180 as well and I did it as u said, it was shown as gen2 , now how would it be benificial for me? how to unlock its langauge options ?

ma1co, 2021-01-03:
@BasitShaikh01 Thank you, everything worked as expected.

The next step is to dump the firmware. Here's how to do that:

* Download the newest [build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) of pmca-console and place in some directory (it is a different version than before).
* Connect the camera.
* Open the command line and navigate to the directory (same as before)
* Run `pmca-console-v0.17-43-g2c24d4b-win.exe updatershell`
* The camera should switch to a black screen. The command line should prompt you for a new command.
* Run the following command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following command to dump the bootloader partitions: `bootloader`
* Run the following command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot` on your directory Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

ma1co, 2021-01-03:
@m-hassan-memon For the PJ630V, please post the full output. For the CX180, please run the script using [0x11010000.zip](https://github.com/ma1co/Sony-PMCA-RE/files/5437971/0x11010000.zip) and post the full output. Thanks.

BasitShaikh01, 2021-01-04:
@ma1co but i don't have nflasha and boot file and I dn't get understand this line

Run the following command to dump the firmware (this will take some time to complete): pull /dev/nflasha

ma1co, 2021-01-04:
@BasitShaikh01 Please read the instructions again. You only need the new version of pmca-console, nothing else. Run `pmca-console-v0.17-43-g2c24d4b-win.exe updatershell` as described above. It should print this:
```
Welcome to the USB debug shell.
>
```
Here, you can enter the next command: `pull /dev/nflasha`. This will create the nflasha file on your computer that you can upload to my dropbox.

BasitShaikh01, 2021-01-04:
@ma1co  updated version of pmca-console has failed to execute script pmca-console
[image]

ma1co, 2021-01-04:
@BasitShaikh01 The error message says it: The camera has to be powered through the AC adapter.

BasitShaikh01, 2021-01-04:
@ma1co camcorder is full charge no AC adapter coneected

BasitShaikh01, 2021-01-04:
@ma1co there is no AC Adapter plug.

ma1co, 2021-01-04:
[There is](https://helpguide.sony.net/gbmig/44520901/v1/gb/contents/02/02/01/01.html) (search for "DC IN jack").

BasitShaikh01, 2021-01-04:
@ma1co yeah i got it but if i connect DC in jack so battery remove or not?

ma1co, 2021-01-04:
You can leave it in.

BasitShaikh01, 2021-01-04:
@ma1co  Sir camera is not Opening after i write this command pmca-console-v0.17-43-g2c24d4b-win.exe updatershell.
[image]

BasitShaikh01, 2021-01-04:
@ma1co what i do?

BasitShaikh01, 2021-01-05:
@ma1co sir camera is ON but language not change. Pmca-console error

BasitShaikh01, 2021-01-08:
@ma1co ?

ma1co, 2021-01-22:
@BasitShaikh01 Sorry for the very late answer. This means that there has been a USB communication error. Make sure that no other programs are open (including things like dropbox) and try a few times.

T-CAME, 2022-09-13:
Hello. I was not able to change the language of SONY HDR-CX535. I really need your help

T-CAME, 2022-09-13:
@ma1co Hello. I was not able to change the language of SONY HDR-CX535. I really need your help

T-CAME, 2022-09-13:
@BasitShaikh01 have you successfully done pj630?

cx670camera, 2022-09-21:
Hi!
I need your help.  I am using camcorder Sony HDR-CX70 japanese language. I want to change to English. I followed your instructions. When it comes to unlocking, I can't open. Can yuu help me unlock it. Thanks!
[image]

m3h3dimithu, 2023-09-27:
Hello @ma1co, Can you help me to change my HDR PJ210 model Handicam's language Japanese to English??