ma1co/OpenMemories-Tweak #359: Change language RX1R

zorbec21, 2019-12-26:
Hello,
I want to change the language of my camera RX1R  from Japanese to English.
I tried pmca-gui V 0.17 and I have this error:

**Get camera info :**
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-RX1R is a camera in MTP mode

Model:              DSC-RX1R
Product code:       0080814200
Serial number:      06016825
Firmware version:   1.00

**Start Tweaking :** 
Querying MTP device
Sony Corporation DSC-RX1R is a camera in MTP mode

Getting device info
Using firmware for model DSC-RX1R
Unknown device: DSC-RX1R

Install App 
Switching to app install mode
Traceback (most recent call last):
  File "C:\projects\sony-pmca-re\pmca-gui.py", line 76, in do
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 274, in installCommand
  File "C:\projects\sony-pmca-re\pmca\commands\usb.py", line 37, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 306, in switchToAppInstaller
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 244, in _sendCommand
  File "C:\projects\sony-pmca-re\pmca\usb\sony.py", line 98, in sendSonyExtCommand
  File "C:\projects\sony-pmca-re\pmca\usb\__init__.py", line 81, in _checkResponse
pmca.usb.InvalidCommandException: MTP error 0x2006

My computer is with Windows 10, I have try with a virtual machine with windows 7, it does not work. Same error MTP 0X2006
I have changed my cables. But I cant update.
Is it possible to update my camera? has anyone the same problem.
Thanks

zorbec21, 2019-12-28:
Problem solved 
With the last version https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts
My RX1R is supported.
And now in French!!!
Thanks

closed 2019-12-28 by zorbec21

jimmij1968, 2022-06-17:
please let us know how did you do it  rx1r from japanese to english
step by step please