ma1co/OpenMemories-Tweak #368: a5000 enable LIVE-VIEW in Sony Imaging Edge Desktop / Remote for PC

SonyILCE, 2020-04-05:
Here is a genius guide how to enable video recording for the android / iOS remote app:
[https://github.com/ma1co/OpenMemories-Tweak/issues/134](url)

It is possible that all the cameras can use live-view in the android-app:
_ILCE-5000, ILCE-5100, ILCE-6000, ILCE-6100, ILCE-6300, ILCE-6400, ILCE-6500, ILCE-6600 and so on_

But only a few cameras support live-view for the 'Sony Imaging Edge Desktop app' for Mac OS / Windows:
official:
_ILCE-6100, ILCE-6400, ILCE-6600 and others_
Source: [https://support.d-imaging.sony.co.jp/app/imagingedge/en/devices/#emount](url)
unofficially it also works with:
_ILCE-6300 and ILCE-6500_

**It would be really nice if the ILCE-5000 could do live-view on Windows/Mac OS.**
Maybe it is possible (via telnet?) because live-view is generally (for mobile app) available.

Has someone a solution or an idea?

domlinson, 2020-04-23:
I've been searching for this too! I've got both an A5000 and a A6000, neither of which have USB Live view enabled... I've been reading through some of the other issues opened on this repo, and i've seen that 'setting' values can be changed over TelNet. 

Does anyone know if there's a lookup table for these setting values? Or if someone has an A6300 dump, so we can compare it to an A6000 dump?

Thanks, 
Dom

domlinson, 2020-04-23:
Just so I don't forget this, using [fwtool.py](https://github.com/ma1co/fwtool.py) - I have unpacked the DUMP.DAT file that came off of my A5000... I then used the print_backup feature of fwtool to further unpack my .BIN file and spat it out into a .TXT file (about 8.4MB)...

So the next step would be to compare the dump to the closest camera that _does_ have the USB live view option enabled...

So I think it would make most sense for me to compare my A6000 to the A6300, as the 6300 does have USB live view.

However... there are a huge number of settings in the .bin file which really makes this difficult. I imagine there will be thousands of differences straight off the bat from the A6000 to the A6300, so how would we determine which is the USB live view?

If the setting was toggleable (ie, the camera menu can turn the feature on/off) we could take 2 dumps from an A6300 with it enabled/disabled and then narrow down our search, however this feature (as far as I know) is just a backend setting, with no way for the user to enable or disable it... I know the Imagine Edge software can show/hide the USB live view feature, but I doubt this changes any settings in the actual camera (but we shouldn't write that off as a possibility).

So...
A) does anyone have an A6300 who'd be able to share some DUMP.DAT files with us?
B) does anyone know if there's a way to toggle the USB Live View feature on/off on an A6300?

Thanks all!

SonyILCE, 2020-04-25:
Hi have an A6300. But what should I do?

domlinson, 2020-04-25:
@SonyILCE in the settings on the camera, is there any toggle or setting around 'Live View' or similar for PC Remote or USB connection? I don't think there will be (apart from the 'USB Mode' selector when you change it to 'PC Remote')... Just seeing if there's a toggle for the USB Live View, or if it's just on by default.

SonyILCE, 2020-04-25:
Don't seems so :/
[image]
[image]
[image]

domlinson, 2020-04-25:
@SonyILCE ahh okay, thanks for trying! I think we need someone with more experience in the firmware to be able to solve this. I tried decompiling some of the firmware files to see if there were any clues about what setting addresses mean what, but the decompilation didn't work for me.

fvollmer, 2020-05-09:
Maybe a starting point (A6000 firmware):
`libObj.so` contains the following strings:
 * `[IMCfg]m_funcInfo.funcLiveStreaming   = %d`
 * `LIVE_STREAMING `
 * `ModelUsbLiveViewToInstance`

The function at `0x0014cbac` uses the string `[IMCfg]m_funcInfo.funcLiveStreaming   = %d` with `%d` the attribute from `0x00cf040b` in `backup.bin`.

I don't know where `ModelUsbLiveViewToInstance` and `LIVE_STREAMING` is used. Maybe someone wants to dig into this.

[image]

queeup, 2020-05-12:
I love to have this on my a6000 with `gphoto2 --capture-movie`

Capobuf, 2020-05-17:
Some news about?

benybkim, 2020-06-16:
I would love this to be enabled for a5100 as well. It would be very nice to be able to use this feature without upgrading my camera to 6x00 series. Any update on this?

reecepounder, 2020-08-12:
Sorry to bump, but I wanted to share my interest as well.

benybkim, 2020-08-30:
> Update: Sony seems to have released an official tool to support this. At least, a5100 is supported. This may be helpful

Would you mind sharing this update? I downloaded Image Edge and updated it but I still am not able to see LIVE-VIEW on my desktop from a5100. 
Thanks!

benybkim, 2020-08-30:
Answering my own question! 
https://support.d-imaging.sony.co.jp/app/webcam/en/download/
New utility is called Imaging Edge Webcam and not part of the Imaging Edge software.

Milad, 2020-09-29:
a5000 is not supported!

pixxelpusher, 2020-10-28:
> Answering my own question!
> https://support.d-imaging.sony.co.jp/app/webcam/en/download/
> New utility is called Imaging Edge Webcam and not part of the Imaging Edge software.

a6000 still not on the list! Would be great to get liveview working in imaging edge desktop (wifi stream too).

cukabeka, 2020-12-09:
> a5000 is not supported!

Yes, can confirm this so far. When Using a mac, no Webcam software shows up anywhere, unfortunately after install. In the download dialog, only the ilc-5100 is mentioned.

sidrichmill, 2020-12-09:
I've been looking for a solution to this on the A6000 for a while! As another possible direction, [Qdslrdashboard](https://dslrdashboard.info/) is able to do a live-view while connected to the A-6000 over wifi. I would really love to be able to use live view with more robust software though. Thanks to anyone chipping away at this problem!

cukabeka, 2020-12-15:
Hi, just a further note regarding the investigation of this issue. What I learned, the ILCE5000 and ILCE5100 have different sensors, with the 5100 having the same as the 6000. 

What I already noticed on the Image Edge Remote App is that I can set the settings and trigger the shutter to make a photo, but there is no live view. When I try to enable it, there's only a spinner shown. I had this issue with 2 macs that I tried the Edge Remote App with. 

Maybe this is related somehow? 

Qdslrdashboard does not connect at all with the ilce5000, although it is mentioned explicitly in the compatibility list. Can someone confirm whether the 5000 does ever transmit a live image to any app?

dennis-fan, 2021-02-21:
> Hi, just a further note regarding the investigation of this issue. What I learned, the ILCE5000 and ILCE5100 have different sensors, with the 5100 having the same as the 6000.
> 
> What I already noticed on the Image Edge Remote App is that I can set the settings and trigger the shutter to make a photo, but there is no live view. When I try to enable it, there's only a spinner shown. I had this issue with 2 macs that I tried the Edge Remote App with.
> 
> Maybe this is related somehow?
> 
> Qdslrdashboard does not connect at all with the ilce5000, although it is mentioned explicitly in the compatibility list. Can someone confirm whether the 5000 does ever transmit a live image to any app?
> 
> Additionally, I found this post [#134 (comment)](https://github.com/ma1co/OpenMemories-Tweak/issues/134#issuecomment-367367134) very enlighting, but could not yet try to rename and connect, but this seems to be related, too, right?

I have tried renaming my A6000 camera to ILCE-6300 but the live view and sony webcam desktop applications for windows still do not work with it.

mungewell, 2021-04-07:
Two things to add to conversation.
1). Sony have a history of disabling features on the A5000 and A6000 which would normally just work, exhibit 1 is the code of the 'SmartRemoteControl' App.
https://github.com/ma1co/Sony-PMCA-RE/issues/67

2). Once hacked they **can** use WiFi connection and 'LiveView' to get a low framerate to PC. I have been using this other project.
https://github.com/Bloodevil/sony_camera_api

Note: the on camera/pre-installed remote app is a bit crappy, make sure that you install the latest one from Sony Playmemories App Store.

saskenuba, 2021-05-05:
> Two things to add to conversation.
> 1). Sony have a history of disabling features on the A5000 and A6000 which would normally just work, exhibit 1 is the code of the 'SmartRemoteControl' App.
> [ma1co/Sony-PMCA-RE#67](https://github.com/ma1co/Sony-PMCA-RE/issues/67)
> 
> 2). Once hacked they **can** use WiFi connection and 'LiveView' to get a low framerate to PC. I have been using this other project.
> [Bloodevil/sony_camera_api](https://github.com/Bloodevil/sony_camera_api)
> 
> Note: the on camera/pre-installed remote app is a bit crappy, make sure that you install the latest one from Sony Playmemories App Store.

Thanks for the tips! I've managed to get a LiveView out of my DSC-rx100m3, after using the authentication.py and then sony_live_preview.py.

The next step is to get it working within OBS software :)

mungewell, 2021-05-05:
Awesome, glad it was of use... BTW there is a branch which includes some undocumented functions (although none of my cameras respond to these):
https://github.com/mungewell/sony_camera_api/blob/undoc2/examples/authenticate.py

wedeservedit, 2021-05-15:
The other alternative I've found is connecting the sony camera (can even do this with my really old qx100) through the imaging edge mobile app on an android phone and screen mirroring it through usb. This is low latency compared to connecting it straight by wifi to pc/mac it seems.

mungewell, 2021-05-19:
@wedeservedit Good suggestion, never thought to use Android screen sharing. There may also be other ways to run the official Sony app on a PC (VM, ChromeApps, etc).

The Python base project only show that it's possible, and could definitely benefit from some optimization.