ma1co/OpenMemories-Tweak #369: Support for DSC-HX50

Gsavas, 2020-04-20:
Hi, looking forward to tweaking this camera, but it seems that it's not supported. When I try to run updatershell: 
`Getting device info
Using firmware for model DSC-HX50

Unknown device: DSC-HX50`
I saw that someone else had already posted a request for HX50V, which is basically the same device minus gps functionality. Here's my guess_firmware output:
`Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Model name: DSC-HX50

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen2_0x31020051_0.dat`

ma1co, 2020-04-21:
The HX50 should already be supported. Are you using the newest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts)?

If you manage to get it to work, I'd be interested in a firmware dump.

Gsavas, 2020-04-21:
I am sorry for wasting your time, I downloaded the build listed at README and I assumed it was the latest release. After trying the one you linked it worked as expected. At your request I dumped the firmware and uploaded it to your dropbox. Thanks for the help!

ma1co, 2020-04-21:
Thank you very much!

closed 2020-04-21 by ma1co