ma1co/OpenMemories-Tweak #378: HDR-MV1 support?

chrometsuba, 2020-06-25:
Any chance support for this camera can be added?  

I have a Japanese version which I am hoping to enable English on.

thanks

chrometsuba, 2020-06-25:
Hoping to find some help, I ran pmca-console guess_firmware -f -x31010000.zip and got the following output:

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Getting device info
Model name: HDR-MV1

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen2_0x31010090_0.dat

I tried to run pmca-console updatershell but it responds with unknown device: HDR-MV1

ma1co, 2020-06-26:
Cool, thank you for your help, the MV1 is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest [build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) of pmca-console.
* Connect the camera.
* Run `pmca-console updatershell`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

chrometsuba, 2020-06-26:
Hi ma1co,

Just dumped and uploaded the files, thanks for the quick update on the pmca app to get this.  You do some fast work my man!!

I haven't had much chance to look into what this camera can do in english, but any chance that there is a hidden menu item to adjust the video data rate to capture higher quality video?

Also, gotta give you some thanks, I used the tweak to remove the time limit on my A72, too bad Sony puts that in there, I would have paid a bit extra to have unlimited out of the box.

ma1co, 2020-06-26:
Thank you for the dump!

chrometsuba, 2020-06-26:
I checked in the PMCA GUI app you just compiled, and the tweak functions work correctly (I now have English) which is great!  Thank you so much for all the work you do on this.

closed 2022-08-17 by ma1co