ma1co/OpenMemories-Tweak #392: I want to unlock HDR-CX670 languages.

johnoh125, 2021-01-08:
I admire your tool.
But I don't know what to do.

There is Cx670 in the list of 'Supported cameras', but it doesn't work.

in Gui,

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Getting device info
Using firmware for model HDR-CX670

Unknown device: HDR-CX670

so, I using '0x31010000.zip'

And This,

d:\downlord>pmca-consol guess_firmware -f 0x31010000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony Camcorder is a camera in mass storage mode

Getting device info
Model name: HDR-CX670

Trying 100 firmware images
Fail: No matching file found

I want to unlock the language.
What should I do?

johnoh125, 2021-01-09:
I tried it 
[https://github.com/ma1co/OpenMemories-Tweak/issues/301](url)

And did it

[image]

[image]

but  tweak not work.... 

[image]

What should I do next?

johnoh125, 2021-01-09:
o, 

I choose number five.

[image]

johnoh125, 2021-01-09:
I tried it

[https://github.com/ma1co/OpenMemories-Tweak/issues/301#issue-370213205](url)

And did it

[image]
[image]

but tweak not work....

[image]
[image]

Um... should I give you the boot file?
I sent the boot file to your dropbox. Please check it.

ma1co, 2021-01-26:
There is a small number of cameras where the protection tweak does not work. The CX670 might be one of them. Can you upload the _nflasha_ file to my dropbox, too? It contains the firmware. The _boot_ file is only the bootloader.

johnoh125, 2021-01-30:
Oh, thank you for your answer.
I have to send nflasha, too.

I just delivered the compressed file to your dropbox.

Thank you.

mklee05, 2021-03-25:
Hello! I bought a Sony HDR-CX670 (Japanese version) . It only provides the Japanese language. Could you please help me unlock the language ,changing to English?

ma1co, 2021-08-12:
@johnoh125 Thank you for the dump.
@mklee05 The HDR-CX670 is supported in pmca-gui using the "Tweaks" tab.