ma1co/OpenMemories-Tweak #394: DSC-WX 200 language unlock

win0809, 2021-02-05:
Hello sir,its nice to know your works,can you help me with unlocking the language of DSC WX 200,its in Japanese. I have tried the console and the gui,i think its not compatible with the updatershell. thanks and more power
[image]

ma1co, 2021-02-06:
The WX200 can probably be added. As a first step, we have to dump its firmware. If you're interested to help, here's what you can do:

* Download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Download [0x31020000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/2624485/0x31020000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x31020000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

win0809, 2021-02-07:
Good day Sir, thanks for the reply,here is what i got from the command:

C:\Users\Win\Desktop\PCMA Console>pmca-console guess_firmware -f 0x31020000.zip
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-WX200 is a camera in MTP mode

Getting device info
Model name: DSC-WX200

Trying 100 firmware images
Firmware update error: Invalid version
Success: Found matching file: gen2_0x31020030_0.dat

ma1co, 2021-02-07:
Thank you, the WX200 is now supported in updatershell.

The next step is to dump the firmware. Here's how to do that:

* Download the newest build of pmca-console.
* Connect the camera.
* Run `pmca-console updatershell`
* Run the following updatershell command to dump the firmware (this will take some time to complete): `pull /dev/nflasha`
* Run the following updatershell command to dump the bootloader partitions: `bootloader`
* Run the following updatershell command to reboot your camera back to normal mode: `exit`
* You should now have files called `nflasha` and `boot` on your computer. Please zip them and upload the file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx).

Thank you for your help!

win0809, 2021-02-07:
Hello sir,i idownloaded the latest console,but im still getting this:

C:\Users\Win\Desktop\P>pmca-console-v0.17-win.exe updatershell
Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying MTP device
Sony Corporation DSC-WX200 is a camera in MTP mode

Getting device info
Using firmware for model DSC-WX200

Unknown device: DSC-WX200

did i do it wrong?

win0809, 2021-02-07:
[image]

ma1co, 2021-02-07:
You need the latest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts) of pmca-console.

win0809, 2021-02-07:
Got it Sir :) I already uploaded the files to your dropbox, have a good day

win0809, 2021-02-08:
Thank you so much for your great work sir,all the tweaks are working,i can now use the camera with ease 🤩. More power to you
[image]

closed 2021-02-08 by win0809