ma1co/OpenMemories-Tweak #396: SLT-A77V Tweaking issue and HDMI question

alexilince, 2021-02-25:
I have the camera Sony SLT-A77V and I have used the app with Tweaking mode. It doesn't work beacouse I can't switch between 50/60 fps. 

Also I have a question for you. Is there any chance to include a hdmi clean output in this camera?

Thank you very much for all your work, it is awesome.

mungewell, 2021-04-07:
The A77 would appear to be on the list of hackable cameras....
```
SLT-A77:
  gen: 2
  model: 0x12030011
```
Can you explain more what doesn't work?

Under the hood the cameras are controlled by a database of settings (withing 'backup.bin'), each of these keys means something to the camera. For the clean HDMI on the A5000 it is key '0x01070a47', which is normally set to '01'. Setting this to '00' turns of the overlay on the output. See #70 for more info.

It is worth seeing whether this performs the same on your camera....

saikacat, 2021-04-10:
> The A77 would appear to be on the list of hackable cameras....
>
> ```
> SLT-A77:
>   gen: 2
>   model: 0x12030011
> ```
>
> Can you explain more what doesn't work?
>
> Under the hood the cameras are controlled by a [...]

Hey mungewell, I have the same camera as well. Since the A77V does not support Wifi connections, I read that the only way I can install tweaks on this camera is through firmware update. I installed the tweaks through the GUI fine, and my camera restarted. I was wondering if you have any experience with dumping the firmware and changing parts of it so I get clean HDMI output? Thanks!

mungewell, 2021-04-10:
The settings for the camera are held in a file 'Backup.bin'. Tweak provides a method to change some (ie Video runtime) and a telnet shell to change others via 'bk' command. You do not need to make FW changes to change the 'Backup.bin' settings.

The SonyRE tools provides a Python script which can be used to connect to a camera via USB _- which it sounds like you have already used_. It places the camera in FW update mode (hence the name of the script) and also has inbuilt 'bk' ability as well as shell access (for nosing around in camera).

I have only done this with Linux, but it may be possible with Windows/Mac. If not you could temporally boot PC into Linux with Ubuntu USB to achieve this if not.... This is me setting the '0x01070a47 key' on my a5000 to turn off the OSD on HDMI output, your camera _might_ support the same key.
```
simon@thevoid:~/Sony-PMCA-RE-github$ sudo python3 pmca-console.py updatershell -m ILCE-5000
No native drivers available
Using drivers libusb-MSC, libusb-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Starting updater shell...

Welcome to the USB debug shell.
Type `help` for the list of supported commands.
Type `exit` to quit.
>bk r 0x01070a47
01
>bk w 0x01070a47 0
Success
>exit
Done
```
_Note: I only have to use the '-m' flag as I have changed the camera's model number, see: https://github.com/ma1co/Sony-PMCA-RE/issues/67_

mungewell, 2021-04-10:
Just to be clear about my FW comment. My a5000 **already** has the FW to turn on/off the OSD on HDMI, just the camera UI does not expose this setting to the user.

Changing the '0x01070a47 key' is done directly and the camera's FW does what it is told ;-)

beng3r, 2021-04-11:
> The settings for the camera are held in a file 'Backup.bin'. Tweak provides a method to change some (ie Video runtime) and a telnet shell to change others via 'bk' command. You do not need to make FW changes to change the 'Backup.bin' settings.
> 
> The SonyRE tools provides a Python script which can be used to connect to a camera via USB _- which it sounds like you have already used_. It places the camera in FW update mode (hence the name of the script) and also has inbuilt 'bk' ability as well as shell access (for nosing around in camera).
> 
> I have only done this with Linux, but it may be possible with Windows/Mac. If not you could temporally boot PC into Linux with Ubuntu USB to achieve this if not.... This is me setting the '0x01070a47 key' on my a5000 to turn off the OSD on HDMI output, your camera _might_ support the same key.
> 
> ```
> simon@thevoid:~/Sony-PMCA-RE-github$ sudo python3 pmca-console.py updatershell -m ILCE-5000
> No native drivers available
> Using drivers libusb-MSC, libusb-MTP
> Looking for Sony devices
> 
> Querying mass storage device
> Sony DSC is a camera in mass storage mode
> 
> Initializing firmware update
> Switching to updater mode
> 
> Waiting for camera to switch...
> Please follow the instructions on the camera screen.
> Initializing firmware update
> Writing firmware
> 0%
> 100%
> Starting updater shell...
> 
> Welcome to the USB debug shell.
> Type `help` for the list of supported commands.
> Type `exit` to quit.
> >bk r 0x01070a47
> 01
> >bk w 0x01070a47 0
> Success
> >exit
> Done
> ```
> 
> _Note: I only have to use the '-m' flag as I have changed the camera's model number, see: [ma1co/Sony-PMCA-RE#67](https://github.com/ma1co/Sony-PMCA-RE/issues/67)_

hi, i have SLT-A58 and i already try this step, i can change the key to 0 and success but didnt work, the hdmi output is still not clean

mungewell, 2021-04-11:
@beng3r Thank you for the confirmation, even if it wasn't what we wanted to hear...

Maybe it would be helpful to clarify what OSD is present on the HDMI output vs the LCD screen and what the menus allow you to change.

@saikacat The act of dumping the firmware is trival, it's more difficult to actually find what/understand how to change.
`simon@thevoid:~/camera_fw$ python3 ~/fwtool-git/fwtool.py unpack -f Update_SLTA77V107.exe -o a77/`

This looks a bit different from the a5000 (which I have been trying to understand). There are references to HDMI and OSD in `libBaseClass.so`, but the `CautionConfig.so` is much simpler than on the a5000/a5100/a65000.

Maybe one of these hints at a setting which can be changed to remove the OSD
```
simon@thevoid:~/camera_fw/a77$ arm-linux-gnueabi-nm -D firmware.tar_unpacked/0700_part_image/dev/nflasha8_unpacked_unpacked/lib/libBaseClass.so | grep -e 'OSD' -e 'HDMI'
0001cdb8 T _ZN17OBJAVBBCBASECLASS5VdfCh10setOSDAreaERKN3VDF26VDF_PRM_INPUT_SET_OSD_AREAE
0001c9b2 T _ZN17OBJAVBBCBASECLASS5VdfCh11setOSDOnOffERKN3VDF28VDF_PRM_INPUT_SET_OSD_ON_OFFE
0001d210 T _ZN17OBJAVBBCBASECLASS5VdfCh12setOSDLayoutERKN3VDF28VDF_PRM_INPUT_SET_OSD_LAYOUTE
0001cc8e T _ZN17OBJAVBBCBASECLASS5VdfCh15setOSDAreaOnOffERKN3VDF33VDF_PRM_INPUT_SET_OSD_AREA_ON_OFFE
0001d9e6 T _ZN17OBJAVBBCBASECLASS5VdfCh9setOSDMixERKN3VDF24VDF_PRM_PATH_SET_OSD_MIXE
00022a22 T _ZN17OBJAVBBCBASECLASS6HdmiCh11getTCOutputERN10HDMI_INFRA27HDMI_PRM_GET_TC_OUTPUT_COMPE
00023064 T _ZN17OBJAVBBCBASECLASS6HdmiCh11set3DFormatERKN10HDMI_INFRA22HDMI_PRM_SET_3D_FORMATE
00022904 T _ZN17OBJAVBBCBASECLASS6HdmiCh11setTCOutputERKN10HDMI_INFRA22HDMI_PRM_SET_TC_OUTPUTE
00022e18 T _ZN17OBJAVBBCBASECLASS6HdmiCh13getDefinitionERN10HDMI_INFRA28HDMI_PRM_GET_DEFINITION_COMPE
000223f6 T _ZN17OBJAVBBCBASECLASS6HdmiCh13getResolutionERN10HDMI_INFRA28HDMI_PRM_GET_RESOLUTION_COMPE
00022b4a T _ZN17OBJAVBBCBASECLASS6HdmiCh13setDefinitionERKN10HDMI_INFRA23HDMI_PRM_SET_DEFINITIONE
000222d6 T _ZN17OBJAVBBCBASECLASS6HdmiCh13setResolutionERKN10HDMI_INFRA23HDMI_PRM_SET_RESOLUTIONE
00022636 T _ZN17OBJAVBBCBASECLASS6HdmiCh15getPremiumPhotoERN10HDMI_INFRA31HDMI_PRM_GET_PREMIUM_PHOTO_COMPE
0002251a T _ZN17OBJAVBBCBASECLASS6HdmiCh15setPremiumPhotoERKN10HDMI_INFRA26HDMI_PRM_SET_PREMIUM_PHOTOE
00023188 T _ZN17OBJAVBBCBASECLASS6HdmiCh18set3DOperatingModeERKN10HDMI_INFRA30HDMI_PRM_SET_3D_OPERATING_MODEE
00022f40 T _ZN17OBJAVBBCBASECLASS6HdmiCh9set3DModeERKN10HDMI_INFRA20HDMI_PRM_SET_3D_MODEE
```

beng3r, 2021-07-01:
> @beng3r Thank you for the confirmation, even if it wasn't what we wanted to hear...
> 
> Maybe it would be helpful to clarify what OSD is present on the HDMI output vs the LCD screen and what the menus allow you to change.
> 

hey, sorry for late reply, the OSD for HDMI output and on the LCD screen are same(green box/Focus). 
menus that allow to change, i think is 'CTRL FOR HDMI'

> I have only done this with Linux, but it may be possible with Windows/Mac. If not you could temporally boot PC into Linux with Ubuntu USB to achieve this if not.... This is me setting the '0x01070a47 key' on my a5000 to turn off the OSD on HDMI output, your camera _might_ support the same key.
> 

i dont think this key support the same for my camera Sony SLT-A58,
do you know how can i start/find the key to disable OSD

mungewell, 2021-07-23:
I threw out some **wild speculation** on a way to 'brute force' which might be the key for this... This might work for the a77 (though not proven on the nex6 yet).
https://github.com/ma1co/Sony-PMCA-RE/issues/249

`viewunified4.so` on the a77 appears to know about the following keys (in the 0x0107xxxx range).
[a77_view4_targets.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6866200/a77_view4_targets.txt)

Speculation is that if the a77 supports this functionality it would be one of the 1 byte keys, factory set to 0x01... there are only 3 to check - so in all likely hood this functionality does not exist :-(

It might be worth poking around though, and maybe seeing if changing the amount of GUI on the camera alters any of these keys.

evilcrusher2, 2022-05-02:
> i dont think this key support the same for my camera Sony SLT-A58, do you know how can i start/find the key to disable OSD

Did you ever find an answer to this? I've seen where this has been asked on [several threads now ](https://github.com/ma1co/Sony-PMCA-RE/issues/235) but there has been zero response to this. 

I imagine There has to be a way to do so. I say this because when I initially hooked up my micro hdmi camera the output was not the 1920x1080p I should have been getting, I was getting 1440 and it appeared to be a clean HDMI output. Maybe I was mistaken. Just seems super odd that when I finally got it to output 1080p, the display won't eliminate the info at the bottom of the screen. 

If I knew what I needed to be looking for on the coding side and how to even see that, I would might be able to solve this if there is a possible solution. Something in the code is telling the screen post that data up while not posting up others.

ofunny, 2023-01-08:
Did anybody succeed on the SLT a58 – I have an old a58 and I would actually like to reuse it for streaming if I would be able to get clean HDMI in 1080p?