ma1co/OpenMemories-Tweak #397: A5100 silent / electronic shutter (feature request)

mo8Zomo0, 2021-03-03:
It would be interesting if cameras like the A5100 (which has a quite loud shutter)
could be silenced (like for timelapses). 
Later A6x00 cameras and some others seem to  support electronic shutter mode,
this would be a great feature for the other cameras as well.

mungewell, 2021-04-10:
I have been trolling through the FW for A5100, and noticed that there is a list of settings in the 'CautionConfig.so' file. These are different between the A5000, A5100 and A6500 (which are the cameras I have looked at) and seem to suggest which features the camera actually supports.... even if the UI does not expose them.

It appears that the A5100 might support 'SilentShutter' as it is listed (the A5000 does not list it)
```
$ grep Silent CautionConfig_a5100.txt 
0028e57c B cmnViewSettingNodeRootSilentShutter
```

I have not yet figured out a way to associate these descriptions to the numerical key used in `Backup.bin`. Once that is known we can set/clear key settings via telnet and `bk.elf`.

If anyone has a camera with SilentShutter and would be able to download `Backup.bin` from the camera with it set/un-set we can look for the change, and identify the key that way.

Hondune, 2022-04-04:
> I have been trolling through the FW for A5100, and noticed that there is a list of settings in the 'CautionConfig.so' file. These are different between the A5000, A5100 and A6500 (which are the cameras I have looked at) and seem to suggest which features the camera actually supports.... even if the UI does not expose them.
> 
> It appears that the A5100 might support 'SilentShutter' as it is listed (the A5000 does not list it - CORRECTION a5000 DOES list it)
> 
> ```
> $ grep Silent CautionConfig_a5100.txt 
> 0028e57c B cmnViewSettingNodeRootSilentShutter
> ```
> 
> I have not yet figured out a way to associate these descriptions to the numerical key used in `Backup.bin`. Once that is known we can set/clear key settings via telnet and `bk.elf`.
> 
> If anyone has a camera with SilentShutter and would be able to download `Backup.bin` from the camera with it set/un-set we can look for the change, and identify the key that way.

Hey mungewell, first i just wanted to thank you for all the responses youve given on these issues. By pieces together various answers from you i was able to figure out how to make changes using pmca console, dump the backup file, etc. which is great.

Did you ever figure out the id for silent shutter? I desperately need it for a project im working on using an a5000. I dont have access to another camera that does have that option in the ui, so im unable to compare backup files. Obviously searching through the a5000 backup file is useless as there is too many settings to manually try them all. I attempted id 0x0060028e (the only one with 0028e in it matching the config text you found) but it doesnt seam to be correct. Any help would be massively appreciated, thanks!

Hondune, 2022-04-09:
> I have been trolling through the FW for A5100, and noticed that there is a list of settings in the 'CautionConfig.so' file. These are different between the A5000, A5100 and A6500 (which are the [...]

@mungewell So I managed to get a local shop let me dump the backup.bin file of a camera that has the silent shutter feature. I dumped it with off and then on two times each. There seams to be a number of settings diffed between the on and off files but non of them appear to have any affect on the a5000. Several of them in my testing auto reverted back though, so potentially it requires changing multiple values at the same time? Here are the backup.bin files if you want to have a look 
@ma1co potentially you have some more insight into this?

[SilentShutterBackupBin.zip](https://github.com/ma1co/OpenMemories-Tweak/files/8456074/SilentShutterBackupBin.zip)

mungewell, 2022-04-12:
I looked at the differences between the '.txt' files with Meld to look for potential keys, in many cases 'on2' did NOT match 'on1'.

Are you 100% sure that the logs were taken/labelled correctly?

[image]
[image]

Hondune, 2022-04-12:
> I looked at the differences between the '.txt' files with Meld to look for potential keys, in many cases 'on2' did NOT match 'on1'.
> 
> Are you 100% sure that the logs were taken/labelled correctly?

I noticed the same thing in the files but I am fairly certain they were done correctly. I was also diffing using meld, there are several settings that are only different in the ON files, several settings that are different in all of the files, and several that are the same between various files regardless of ON or OFF (but seam to change in each file). I was however in a shop and trying to hurry as to not be in the way of other customers, its possible i messed one up (but this is also why i dumped several). My method was to turn on the camera, navigate to the silent shutter option, then turn the camera off (my thought being this would call the bk sync function, i have no idea if this is necessary, this also saved the last setting as silent shutter so i didnt have to mess with anything else to change it for the dumps). I then turned the camera on, plugged it in, dumped the settings, and turned it back off. I then turned it on, changed the setting, turned off to bk sync, and repeated the above. I didnt touch anything else or even navigate the menu at all during these dumps in hopes of minimizing the amount of changes in each dump (didnt seam to matter i suppose)

mungewell, 2022-04-13:
I wasn't able to find a key which prevented the shutter on my A5000. But then I found:
https://github.com/ma1co/Sony-PMCA-RE/issues/193

Am I really hearing a shutter, or a recording of a shutter? _BTW same files exist in A5100 FW._

Hondune, 2022-04-13:
> I wasn't able to find a key which prevented the shutter on my A5000. But then I found: [ma1co/Sony-PMCA-RE#193](https://github.com/ma1co/Sony-PMCA-RE/issues/193)
> 
> Am I really hearing a shutter, or a recording of a shutter? _BTW same files exist in A5100 FW._

You can feel and see the physical shutter (take a lens off and watch it while you take a photo)

I'm not sure about the 5100 but the 5000 only has a rear curtain shutter, the front curtain is electronic. So when you start the photo the camera makes a fake noise to make up for the lack of mechanical noise, but at the end of the photo you're hearing the real shutter (very easy to differentiate the two with a long exposure). I did all my tests with a lens off and watching to make sure the physical shutter was moving.

The sound doesn't bother me, but I have a project where I need the internals of the camera WITHOUT the shutter unit attached. The firmware throws a camera error when the shutter unit isn't plugged in (after succesfully taking and saving the picture which makes it so frustrating) and I was hoping this would get around that. Theres gotta be a way to get it working, or some other work around

Hondune, 2022-04-13:
Something else interesting ive found. Under ID 0x00a10003 in SilentOff2.txt there seams to be some text relating to settings:

```
ff ff ff ff 53 4d 49 4c 45 5f 53 48 55 54 54 45  ....SMILE_SHUTTE
  52 5f 4d 4f 44 45 00 00 28 00 00 00 38 00 00 00  R_MODE..(...8...
  78 ac 19 00 01 00 00 00 02 00 00 00 02 00 00 00  x...............
  10 01 07 01 d4 05 1a 00 1e 00 00 00 00 00 00 00  ................
  03 00 00 00 03 00 00 00 03 00 00 00 00 00 00 00  ................
  00 00 00 00 f1 00 00 00 40 c4 19 00 60 c2 3a 04  ........@...`.:.
  02 00 00 00 02 00 00 00 00 00 00 00 5c ad 19 00  ............\...
  1f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 d8 fe 19 00 b9 00 00 00  ................
  40 c4 19 00 60 c2 3a 04 ff ff ff ff 50 53 45 55  @...`.:.....PSEU
  44 4f 5f 52 45 43 5f 4d 20 00 00 00 38 00 00 00  DO_REC_M....8...
  10 ad 19 00 01 00 00 00 02 00 00 00 02 00 00 00  ................
  12 01 07 01 b4 ad 19 00 20 00 00 00 00 00 00 00  ................
  04 00 00 00 04 00 00 00 04 00 00 00 00 00 00 00  ................
  c0 c2 3a 04 61 00 00 00 40 c4 19 00 60 c2 3a 04  ..:.a...@...`.:.
  ff ff ff ff 53 4f 46 54 5f 53 4b 49 4e 5f 45 46  ....SOFT_SKIN_EF
  46 45 43 54 5f 4c 45 56 28 00 00 00 38 00 00 00  FECT_LEV(...8...
  68 ad 19 00 01 00 00 00 02 00 00 00 02 00 00 00  h...............
  00 00 00 00 54 c4 19 00 21 00 00 00 00 00 00 00  ....T...!.......
```

Is there potentially more information in the .bin files that isnt being brought out by the .txt converter?

mungewell, 2022-04-14:
It appears that `0x0107001f` is 'USB Mode'
```
id=0x0107001f, size=0x0001, attr=0x40:
 01 - Auto
 02 - Mass Storage
 03 - MTP
 04 - PC Remote
```

mungewell, 2022-04-14:
Changing key `bk w 0x01070b2c 02` affects the way the camera takes the shot, but still has (real) shutter at end. After blanking screen for shot it does NOT display 'processing'...

Hondune, 2022-04-14:
> It appears that `0x0107001f` is 'USB Mode'
Interesting, I didnt change this so it must have been set to "auto" and was changing itself

> Changing key `bk w 0x01070b2c 02` affects the way the camera takes the shot, but still has (real) shutter at end. After blanking screen for shot it does NOT display 'processing'...
My a5000 doesnt display processing to begin with, though i have already disable long exposure noise reduction (ID 0x01070108) so maybe thats why?

ID 0x01070b2c exists on the a5000 (cant seam to find what it does though) but does NOT exist on the nex-5n

I have also identified ID 0x0107086e  as the electronic front curtain shutter setting. 01 is on (doesnt use mechanical front curtain) and 02 is off (does use mechanical front curtain). Setting this to other values like 00 or 03 automatically reverts it back to 01.

mungewell, 2022-04-15:
I was taking photos in Shutter Priority, 5s exposure with the lens cap on... call me crazy if you like.

Hondune, 2022-04-15:
> I was taking photos in Shutter Priority, 5s exposure with the lens cap on... call me crazy if you like.

at 5s the "processing" would definitely be the long exposure noise reduction. Very strange that 0x01070b2c would change that AND 0x01070108 changes it. I can confirm that 0x01070108 is the long exposure noise reduction setting in the menu, so 0x01070b2c  must be another setting that will also turn off LENR.

Long exposure noise reduction is not available when using silent shutter, so it stands to reason that 0x01070b2c is possibly the setting for silent shutter and it just doesnt work on these cameras for some reason which would be a big bummer. Although it would be odd that it would still exist but just not do anything, whereas it doesnt exist at all on the nex-5n

mungewell, 2022-04-15:
So is LENR the "star eater" that people where complaining about? Lot's of people looking for a way to turn that off...
https://github.com/ma1co/OpenMemories-Tweak/issues/235
https://github.com/ma1co/OpenMemories-Tweak/issues/267

Hondune, 2022-04-15:
> So is LENR the "star eater" that people where complaining about? Lot's of people looking for a way to turn that off... #235 #267

Almost all of sony cameras have the option to turn off LENR in the menu. Only the a5000 and a5100 don't for some reason (even the older nex cameras have it).

The star eater issue is with the actual raw files, sony does some weird sensor noise reduction baked into the raw and it has a habit of removing low power stars which it considers to be sensor hot spots or noise.

tonghuanming, 2023-02-13:
> > I have been trolling through the FW for A5100, and noticed that there is a list of settings in the 'CautionConfig.so' file. These are different between the A5000, A5100 and A6500 (which are the cameras I have looked at) and seem to suggest which features the camera actually supports.... even if the UI does not expose them.
> > It appears that the A5100 might support 'SilentShutter' as it is listed (the A5000 does not list it - CORRECTION a5000 DOES list it)
> > ```
> > $ grep Silent CautionConfig_a5100.txt 
> > 0028e57c B cmnViewSettingNodeRootSilentShutter
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > I have not yet figured out a way to associate these descriptions to the numerical key used in `Backup.bin`. Once that is known we can set/clear key settings via telnet and `bk.elf`.
> > If anyone has a camera with SilentShutter and would be able to download `Backup.bin` from the camera with it set/un-set we can look for the change, and identify the key that way.
> 
> 嘿，芒格威尔，首先我只想感谢你对这些问题的所有回应。通过将您的各种答案拼凑在一起，我能够弄清楚如何使用 pmca控制台进行更改，转储备份文件等，这很棒。
> 
> 你有没有弄清楚静音快门的id？我迫切需要它来完成一个正在使用 a5000 的项目。我无法访问在 ui 中具有该选项的其他相机，因此我无法比较备份文件。显然，搜索a5000备份文件是没有用的，因为有太多的设置可以手动尝试它们。我尝试了 id 0x0060028e（唯一一个与您找到的配置文本匹配的 0028e 的 <>e），但它不正确。任何帮助将不胜感激，谢谢！

TRY THIS 0x1070b2c 

I FIND:  FUN_003cd430(pvVar1,0x1070b2c,0,"SILENT_SHUTTER",0x42);

YOU CAN SET IT TO "1"

dekifx, 2024-03-27:
> > > I have been trolling through the FW for A5100, and noticed that there is a list of settings in the 'CautionConfig.so' file. These are different between the A5000, A5100 and A6500 (which are the cameras I have looked at) and seem to suggest which features the camera actually supports.... even if the UI does not expose them.
> > > It appears that the A5100 might support 'SilentShutter' as it is listed (the A5000 does not list it - CORRECTION a5000 DOES list it)
> > > ```
> > > $ grep Silent CautionConfig_a5100.txt 
> > > 0028e57c B cmnViewSettingNodeRootSilentShutter
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > I have not yet figured out a way to associate these descriptions to the numerical key used in `Backup.bin`. Once that is known we can set/clear key settings via telnet and `bk.elf`.
> > > If anyone has a camera with SilentShutter and would be able to download `Backup.bin` from the camera with it set/un-set we can look for the change, and identify the key that way.
> 
> TRY THIS 0x1070b2c
> 
> I FIND: FUN_003cd430(pvVar1,0x1070b2c,0,"SILENT_SHUTTER",0x42);
> 
> YOU CAN SET IT TO "1"

Hey! It's nice to find you've found a new discovery. How are the commands that I would need to test it? And the command to revert it back, I'm very new to sony camera tweaking/modding so a guide will be appreciated! Thanks..