ma1co/OpenMemories-Tweak #398: A5100 usb "power" (feature request)

mo8Zomo0, 2021-03-03:
It would be great if for cameras like the A5100 the usb mode could be (temporary)
disabled. 
By today if the camera is powered on and I connect a USB power it goes into USB mode
and it can not shoot anymore.

* best would be if charging / powering while connected and still being able to operate the camera would be possible
* or at least the USB connection could be ignored so even if powered on and connected you are still able to use the camera functions

leoleo-s, 2021-03-27:
A “kinda” solution for you would be use a dummy battery.

orestesgaolin, 2021-04-22:
I just decided to buy cheap dummy battery that has its own power supply for around 25 EUR like [this one](https://www.amazon.de/dp/B00T6391ZC/ref=pe_3044161_185740101_TE_item) and it's been working great since November

arkadiusz-wieczorek, 2023-02-21:
There are USB cables without data wire, have you tried to check? In this way I figured that how to power my Sony DSC-HX20V through USB without going to USB mode.