ma1co/OpenMemories-Tweak #400: Sony ILCE-7/R/S Mark II - Solve shutter problems

MarkKrowd, 2021-03-04:
Second generation of ILCE-7 have a huge conception default. Sony has placed an IR optocoupler for the front curtain right in front of the sensor, being a strong light leak and a big problem for astrophotography.
The idea here would be to find a setting for turning this LED off when using Silent Shutter (basically turning the mecanical shutter off).
If this is successfull, this would be great to find a solution for using silent shutter longer than 30s (now you can't go in BULB when silent shooting).
With these 2 improvements, the ILCE-7/R/S/ Mark II would finally be really suitable for astrophotography.

Informations about the problem: 
https://www.cloudynights.com/topic/488461-is-sony-really-alpha/page-62
30s dark ISO 3200 silent shutter ON (NO REASON FOR THIS F***ING LED TO BE TURNED ON!)
[image]

If needed, I'm available to search this setting if I know how to process.

Thanks!

kishorpatel82, 2021-03-13:
This problem has ruined my entire Timelapse stack : https://www.picxy.com/video/milkyway-timelapse-at-lrk-gujarat-38794

mungewell, 2021-04-13:
@MarkKrowd if my understanding/assumptions are correct, these at the key names that the a7Rmk2 knows about... I don't know yet know of a way to associate them with the numerical values used in `Backup.bin`.

`simon@thevoid:~/camera_fw/a7Rmk2$ arm-linux-gnueabi-nm -D ./firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/lib/CautionConfig.so | grep cmnViewSettingNodeRoot | sort > ../CautionConfig_a7Rmk2.txt`
[CautionConfig_a7Rmk2.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6307139/CautionConfig_a7Rmk2.txt)

Do you see anything in there which might be the feature/control you are looking for?

Since you have a camera which supports SilentShutter, can you download a copy of `Backup.bin` with and without it enabled.... this would help for #397?

mungewell, 2021-04-14:
Kind of off topic - read through thread, it's amazing how much motivation these guys have... I'd be really scared opening up such an expensive camera.

I presume the intent of the IR emitter/detector and vane wheel is to measure the speed of the motor, or even the position of the shutter vanes. Both of which are somewhat useless if the shutter is being held open.

I'd be more interested on whether this IR led is turning on/off with various modes/actions on the camera. That might help identify how it is controlled, probably a GPIO on the processor and then where in the software it is done. That would give some (little) hope that that part of the FW can be NOP'ed so that it never turns on during Bulb/SilentShutter.

mungewell, 2021-04-14:
Also, you mentioned that Bulb is not possible during SilentShutter - do 'we' know where that limitation is enforced? Is there perhaps a list of allowable ShutterSpeeds which could be altered to enable Bulb?

mungewell, 2021-04-14:
Had a look to see if I could find the 'ShutterSpeed' in the `Backup.bin` of my a5000, not totally convinced but it might be this key

```
simon@thevoid:~/sony_video_settings/a5000$ grep -A 1 0x00030081 Backup*.txt
Backup_250th.bin.txt:id=0x00030081, size=0x0060, attr=0x68:
Backup_250th.bin.txt-  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
--
Backup_4000th.bin.txt:id=0x00030081, size=0x0060, attr=0x68:
Backup_4000th.bin.txt-  00 00 00 00 00 00 00 1c 00 00 00 00 00 00 00 00  ................
--
Backup_500th.bin.txt:id=0x00030081, size=0x0060, attr=0x68:
Backup_500th.bin.txt-  00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00  ................
--
Backup_bulb.bin.txt:id=0x00030081, size=0x0060, attr=0x68:
Backup_bulb.bin.txt-  00 00 00 00 00 00 ab 0a 00 00 00 00 00 00 00 00  ................
```

You could perhaps manually change your ShutterSpeed and see if you get similar changes. If so you may be able to set 'Bulb' event though the UI doesn't allow it.

Note: If you write values back note that it's quite a large key, and you should probably set all the bytes.
```
id=0x00030081, size=0x0060, attr=0x68:
  00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
reset data:
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

MarkKrowd, 2021-04-14:
> Kind of off topic - read through thread, it's amazing how much motivation these guys have... I'd be really scared opening up such an expensive camera.
>
> I presume the intent of the IR [...]

The LED is ALWAYS turned on... I can understand that it's good to always know where the shutter is, but when you know that it is leaking IR, you should turn it ON only when necessary (example: NOT WHEN POSING!)

MarkKrowd, 2021-04-14:
> Also, you mentioned that Bulb is not possible during SilentShutter - do 'we' know where that limitation is enforced? Is there perhaps a list of allowable ShutterSpeeds which could be altered to enable Bulb?

I had found another thread (I haven't the source anymore) saying that it's a physical limitation about the sensor's buffer used in silent... So you physically can't shoot longer than something like 32s in silent

MarkKrowd, 2021-04-14:
> Had a look to see if I could find the 'ShutterSpeed' in the `Backup.bin` of my a5000, not totally convinced but it might be this key
>
> ```
> simon@thevoid:~/sony_video_settings/a5000$ grep -A 1 [...]

Yeah, someone should just tell me how to properly extract these logs but now I really don't have time, I'm finishing my Bachelor so I'll come back asap to ask some help...

mungewell, 2021-04-15:
If you are using the on camera app 'Tweak' you can use the last tab to enable WiFi and Telnet, once done you can telnet into the camera from a PC to get shell access and then run the `bk.elf` command. Read/Write can be done easily, but listing the whole contents with `bk.elf l` requires you to grab the screen contents into a log file.

I can also confirm that `0x00030081` is the key for setting the Shutter speed, don't know why there is so many bytes/zeros in it. The speed is represented by 2 bytes; which I believe is a non linear value, a change in 1 of the right byte means doubling/halving the speed and the left byte is the fractional.
```

Sets shutter speed to 1/80th
bk.elf w 00030081 00 00 00 00 00 00 55 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 16 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/250th
bk.elf w 00030081 00 00 00 00 00 00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/320th
bk.elf w 00030081 00 00 00 00 00 00 55 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/400th
bk.elf w 00030081 00 00 00 00 00 00 AB 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AB 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/500th
bk.elf w 00030081 00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/1000th
bk.elf w 00030081 00 00 00 00 00 00 00 1a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to 1/4000th
bk.elf w 00030081 00 00 00 00 00 00 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter to 15s
bk.elf w 00030081 00 00 00 00 00 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter to 30s
bk.elf w 00030081 00 00 00 00 00 00 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Sets shutter speed to BULB
bk.elf w 00030081 00 00 00 00 00 00 AB 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AB 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Have to send at least to the last no-zero, ie.
bk.elf w 00030081 00 00 00 00 00 00 AB 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AB 0A
```

xiCO2k, 2021-10-21:
Thanks @mungewell, can you share with me, how I can make the backup from my A5000? Thanks a lot.

Hondune, 2022-04-11:
I have dumped the bin files for silent shooting on/off here https://github.com/ma1co/OpenMemories-Tweak/issues/397 but have been unable thus far to figure out which setting(s) need to be changed to get this working on other cameras. I can see references to silent shooting settings in the dumped firmware all the way down to the a5000 so i think it should be possible. Maybe someone else will have better luck