ma1co/OpenMemories-Tweak #401: HDR PJ670 Mass Storage Error

davidjimmy, 2021-03-15:
Hello ma1co,
I want to change to language settings from Japanese to English. 
The camcorder is HDR PJ670
Per your instruction, I downloaded and ran pmca-gui v0.17. The camcorder is connected via USB.  Laptop is running Windows 10
The GUI successfully identifies the camera info. 
However when I install the OpenMemories: Tweak app, it says
Using driver Windows-MSC, Windows-MTP
Sony camdorder is a camera in mass storage mode.

InvalidCommandException: Mass storage error

I tried power down and up the camcorder several times, powered up and down my laptop several times without any luck.
Also there are no other USB devices attached to the computer.
I'm stuck . Please help !!!!

davidjimmy, 2021-04-04:
Any help is appreciated. I'm still stuck. Need the fix real bad

ma1co, 2021-08-12:
Use the "Tweaks" tab in the newest [development build](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).

closed 2021-08-12 by ma1co