ma1co/OpenMemories-Tweak #403: Is it possible to install the tweaks on Sony a3000?

leoleo-s, 2021-03-24:
I have an a3000 witch i use mostly for YouTube videos and it’s a pain to only records 29 min and 51 secs. 
I tried to install the tweak several times with no luck.
Is it possible on the a3000 or is there some fw limitation?

CyberUSB2, 2021-05-08:
If you're camera can't install apps, then instead use [PMCA-RE](https://github.com/ma1co/Sony-PMCA-RE/). Which supports Sony camera's from 2007 onwards.

ma1co, 2021-06-03:
The A3000 can probably be added. As a first step, we have to dump its firmware. If you're interested to help, here's what you can do:

* If you're on windows, download the newest development build of pmca-console [here](https://ci.appveyor.com/project/ma1co/sony-pmca-re/build/artifacts).
* Download [0x22030000.zip](https://github.com/ma1co/OpenMemories-Tweak/files/6591467/0x22030000.zip) and place it in the same folder (no need to unpack). It contains a bunch of firmware images which might match your camera.
* Power on your camera and connect it over USB to your computer.
* Open a command prompt and run the following command: `pmca-console guess_firmware -f 0x22030000.zip`
* With a bit of luck, the script should find the model id of your camera.

Let me know if you need any help.

leoleo-s, 2021-06-06:
> The A3000 can probably be added. As a first step, we have to dump its firmware. If you're interested to help, here's what you can do:
>
> * If you're on windows, download the newest development build [...]

I’m actually a Mac user, but i do have a Windows laptop laying around. I’ve tried the pmca-console of the link you provided, but it consistently crashes every single time as soon as it opens. My laptop runs on the newest windows 10 version.

ma1co, 2021-06-18:
@leoleo-s You have to run the program in the windows command line. Place the .exe file and the .zip file in the same directory, [open the command line](https://www.computerhope.com/issues/chusedos.htm), run `cd C:\path\to\the\directory` and then run `pmca-console-v0.17-51-g52f5c6f-win.exe guess_firmware -f 0x22030000.zip`

Adnan-Kapidzic, 2024-09-18:
Hi Probably few years too late, but i own this camera and i could help if you are still intrested. :)