ma1co/OpenMemories-Tweak #404: RX100M5 - turn off screen option

marcodagi, 2021-04-06:
Is it possible to add an option(s) to turn off a screen without turning off the camera? I found that newer models have option to turn it off after custom time. This is something good to have (or even more if this kind of thing is possible at all)

CyberUSB2, 2021-05-08:
In order to solve this, you would need to do the following steps:

1. Extract the [backup.bin](https://github.com/ma1co/fwtool.py/issues/6) file from the newer camera ( by using [pmca-console.exe](https://github.com/ma1co/Sony-PMCA-RE/releases) pull file command)
2. Change the setting for turning the screen off (dont change any other settings)
3. Extract the backup.bin file again from the newer camera
4. Compare the 2 extracted backup.bin files and use "diff" software to see which setting has changed
5. Step 4 should have revealed the backup param ID for turning the screen off
6. Use PMCA-console.exe interactive updater shell to manually set that backup param ID on the older camera.
7. Operate your older camera to see if the newer setting has an affect..

dsol1980, 2023-10-01:
Anyone found this missing parameter for older cams?  if someone discovered the parameter, would this work across other Sony models do you think?