ma1co/OpenMemories-Tweak #408: DSC-HX60/HX60V Turn off HDMI Info

Kalminir, 2021-06-24:
Hi, is there a way to be able to turn off HDMI info on a HX60V? Firmware version 2.10

Nalsai, 2021-07-03:
I recently found this: https://there.oughta.be/a/way-to-use-a-sony-alpha-as-a-webcam#getting-rid-of-the-user-interface
You can try if it works.

Kalminir, 2021-07-04:
Tried it but its for a diffrent model. So i ended up losing the functionality that turns the screen off when the hdmi is plugged in lol.

mungewell, 2021-07-23:
By 'turning off info', I assume that you are looking for a 'clean HDMI output' (without overlay graphics).

It should/might be as simple as setting the `0x01070a47` key... that's how it works on the a5000. The HX60 appears to know about this key, so it's worth a try.

My wife has the HX60, I'll give it a go at the weekend.

Kalminir, 2021-07-23:
> By 'turning off info', I assume that you are looking for a 'clean HDMI output' (without overlay graphics).
>
> It should/might be as simple as setting the `0x01070a47` key... that's how it works on the [...]

Yeah that’s the one i already tried. Ending up having live feed on the camera screen and HDMI :P Usually it turns off automatically when a HDMI is connected. So, the 0x01070a47 is at least doing something with the screen. But i might also just be a total bot so all help is appreciated :)

mungewell, 2021-07-25:
Unfortunately my wife left the HX60 at work, I poke around a little with my a5000 and found that the behaviour is somewhat similar - in that the known key is not really turning off the info, but is controlling which screen the recording is heading to. 

Like your HX60; with '01' set the a5000's internal monitor is blanked, and the 'info' display is sent to external monitor. With '00' set the external monitor shows simultaneously as the internal monitor, just that it has no info items displayed. _Though also note that playback of files ALWAYS blanks internal monitor and displays with info on external._

We are assuming that this is the 'NoOsdOutputThroughHdmi' key, but that may not be correct.
```
id=0x01070a47, size=0x0001, attr=0x40:
  00 - internal screen is NOT blanked, external monitor is now clear of info
reset data:
  01 - internal screen IS blanked and display info shown on external monitor.
```

It could be that when the HX60 is display to both internal and external, it does not have the capabilities/code to display different content - the a5000 obviously can...

I also found the key that controls what content is shown, although there was not a 'none' value. Maybe the HX60 also supports this, and might have other values.
```
id=0x010706cb, size=0x0001, attr=0x40:
  03 - level 1 - just speed, Fno, exposure, iso (mode and battery times out and disappears)
  00 - level 2 - above + bar graphs + mode/battery stays
reset data:
  01 - level 3 - all data displayed (different iset for different modes - movie vs I-Auto, etc)

  02, 04 and 05 same as 00
```

Kalminir, 2021-07-25:
Thank you for the reply :D I will also try some more with this new info

Kalminir, 2021-07-28:
> Unfortunately my wife left the HX60 at work, I poke around a little with my a5000 and found that the behaviour is somewhat similar - in that the known key is not really turning off the info, but is controlling which screen the recording is heading to.
> 
> Like your HX60; with '01' set the a5000's internal monitor is blanked, and the 'info' display is sent to external monitor. With '00' set the external monitor shows simultaneously as the internal monitor, just that it has no info items displayed. _Though also note that playback of files ALWAYS blanks internal monitor and displays with info on external._
> 
> We are assuming that this is the 'NoOsdOutputThroughHdmi' key, but that may not be correct.
> 
> ```
> id=0x01070a47, size=0x0001, attr=0x40:
>   00 - internal screen is NOT blanked, external monitor is now clear of info
> reset data:
>   01 - internal screen IS blanked and display info shown on external monitor.
> ```
> 
> It could be that when the HX60 is displaying to both internal and external, it does not have the capabilities/code to display different content - the a5000 obviously can...
> 
> I also found the key that controls what 'info' is shown, although there was not a 'none' value. Maybe the HX60 also supports this, and might have other values.
> 
> ```
> id=0x010706cb, size=0x0001, attr=0x40:
>   03 - level 1 - just speed, Fno, exposure, iso (mode and battery times out and disappears)
>   00 - level 2 - above + bar graphs + mode/battery stays
> reset data:
>   01 - level 3 - all data displayed (different iset for different modes - movie vs I-Auto, etc)
> 
>   02, 04 and 05 same as 00
> ```

Tried some more but with no luck

GerardSD19, 2023-01-03:
Hello! Did you find any solution? Thank you!

arkadiusz-wieczorek, 2023-02-21:
I don't know too much about DSC-XH60V, but I have solution for DSC-HX20V, and maybe it will be reproducible for that model.

You have to go to a movie mode, then click a menu button and change following settings:

- Movie shutting scene: **High Sensitivity** or **Beach** (other modes produce overexposed image or have trouble with right focus on face). In the case of the beach mode, you have to play with EV and remember to set it correctly each time when you turn on/off the camera
- Movie Smile Shutter: **OFF**
- Smile Detection Sensitivity: **LOW**
- Face Detection: **OFF**

After changes, I press a shutter button and click few times a display button to select Display Setting to **OFF**, and now I have a clean HDMI output.

GerardSD19, 2023-02-21:
I can't find the options. Could you record a video please? Thank you!

arkadiusz-wieczorek, 2023-02-27:
Take a look at the instructions of the DSC-HX20V and DSC-XH60V, they should be similar.

Gabriel44200, 2024-09-30:
Any solution?