ma1co/OpenMemories-Tweak #409: Reverting back A7 model number - enabling live-view Sony Imaging App

zenoobs, 2021-06-29:
> > In case someone wants a full guide on how to change the model of Sony A6000 (based on @mungewell's and @syntaxer's post):
> > 
> > 1. Install the **OpenMemories: Tweak** app on your camera and go to the **Protection** tab and select **Unlock protected settings**.
> > 2. Run the **pmca-console** through the command line / terminal with the flag **updatershell** (i.e. `pmca-console-v0.17-win.exe updatershell`).
> > 3. Then change the model name from **ILCE-6000** to **ILCE-6001** with the following command: `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 31 00 00 00 00 00 00 00`.
> > 
> > If you want to **reset** the model name to **ILCE-6000** just write `bk w 0x003e0005 49 4c 43 45 2d 36 30 30 30 00 00 00 00 00 00 00`.
> 
> Thnx, do this with A7, just rename to 7S: bk w 0x003e0005 49 4c 43 45 2d 37 53 00 00 00 00 00 00 00 00 00

Hi, thanks for sharing this, this has work wonders for me. Just another favour, if you can tell me how to revert back to the normal A7 firmware and all.

_Originally posted by @zenoobs in https://github.com/ma1co/OpenMemories-Tweak/issues/134#issuecomment-643584282_

zenoobs, 2021-06-29:
Anyone know the original model number for Sony A7 mark 1? 

I changed it with this code to A7S, but now I want to revert it back to A7.

Any help would be appreciated, cheers.

Kreteen, 2021-08-06:
> Anyone know the original model number for Sony A7 mark 1?
> 
> I changed it with this code to A7S, but now I want to revert it back to A7.
> 
> Any help would be appreciated, cheers.

Just use some ASCII to Hexdec conventer.

ILCE-7:
`49 4c 43 45 2d 37`

closed 2021-08-12 by ma1co