ma1co/OpenMemories-Tweak #410: "OpenMemories was built for an older version of android", can't use on android 10

kubast2, 2021-07-02:


Nalsai, 2021-07-03:
You install the apk on your camera through [pmca](https://github.com/ma1co/Sony-PMCA-RE) to tweak it's settings from the camera directly. 
If your camera actually runs Android 10 it would be helpful to know what model it is.

kubast2, 2021-07-10:
> If your camera actually runs Android 10 it would be helpful to know what model it is.

Wait we really as species put java and android framework on a (lensless) camera in 2013?
Someone told me this will work with Sony Nex-5R

I just were searching for the firmware for custom Sony Nex-5R firmware or at least decrypted firmware, with a clear way on whether or not(and how to) sign/checksum edit the firmware, so that I could just slam in openssh with sftp server compiled so that I can move video files to my phone/computer over wifi.

closed 2021-08-12 by ma1co